maximkulkin/esp32-homekit-camera #8: Wrover Kit - SCCB_Write failed

GermanSheepDog, 2019-02-22:
Hello Maxim,

I used a Wrover Kit camera and if I started the monitor, I got this error:
```
WiFI ready
I (4821) esp32_camera: Free heap: 230212
I (4821) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (4831) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
SCCB_Write [ff]=01 failed
SCCB_Write [12]=80 failed
E (4871) camera: Camera probe failed with error 0x20001
E (4871) esp32_camera: Camera init failed with error 0x20001
>>> HomeKit: Starting server
```
What can I do?

HomeACcessoryKid, 2019-02-23:
You should find out the mapping between the gpio-pins and the camera function pins.

This is what I found in a camera app.
it should be a good start, but no guarantees ;-)

#if CONFIG_CAMERA_MODEL_WROVER_KIT
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    21
#define SIOD_GPIO_NUM    26
#define SIOC_GPIO_NUM    27

#define Y9_GPIO_NUM      35
#define Y8_GPIO_NUM      34
#define Y7_GPIO_NUM      39
#define Y6_GPIO_NUM      36
#define Y5_GPIO_NUM      19
#define Y4_GPIO_NUM      18
#define Y3_GPIO_NUM       5
#define Y2_GPIO_NUM       4
#define VSYNC_GPIO_NUM   25
#define HREF_GPIO_NUM    23
#define PCLK_GPIO_NUM    22

GermanSheepDog, 2019-02-24:
> Antworten an: maximkulkin/esp32-homekit-camera <reply@reply.github.com>

Hello Maxim,

 

for WROVER_Kit from TTgo (exactly: TTGO T-Kamera ESP32 WROVER & PSRAM Kamera Modul ESP32-WROVER-B OV2640 Kamera Modul 0,96 OLED) this worked for me:

 

#if CONFIG_CAMERA_MODEL_WROVER_KIT

#define CAMERA_PIN_PWDN   26

#define CAMERA_PIN_RESET  -1

#define CAMERA_PIN_SIOD   13

#define CAMERA_PIN_SIOC   12

#define CAMERA_PIN_XCLK   32

 

#define CAMERA_PIN_D0      5

#define CAMERA_PIN_D1     14

#define CAMERA_PIN_D2      4

#define CAMERA_PIN_D3     15

#define CAMERA_PIN_D4     18

#define CAMERA_PIN_D5     23

#define CAMERA_PIN_D6     36

#define CAMERA_PIN_D7     39

#define CAMERA_PIN_VSYNC  27

#define CAMERA_PIN_HREF   25

#define CAMERA_PIN_PCLK   19

 

Bye,

Uwe

closed 2019-04-22 by maximkulkin