maximkulkin/esp32-homekit-camera #13: Pairing Code?

KissmanCZ, 2019-05-10:
Hello Maxim, I just flashed your FW to my ESP32 and I am stucked on this:

`>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: AD:B3:70:E8:D9:36
>>> HomeKit: Configuring mDNS
`

I can see cam in my HomeKit, but I am not able to add, becuse I am missing pairing code. I think that boot of my ESP32 is not complete?

beatpaul, 2019-05-10:
Same problem for me?

KissmanCZ, 2019-05-10:
Enabled homekit debug mode:

```
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: AD:B3:70:E8:D9:36
>>> HomeKit: Configuring mDNS
>>> homekit_run_server: Staring HTTP server

```

KissmanCZ, 2019-05-10:
Ok, tried 11111111 code and cam is connected now :)

closed 2019-05-10 by KissmanCZ