maximkulkin/esp32-homekit-camera #15: Not able to compile

AchimPieters, 2019-05-18:
Hi Maxim,

After installing your dockers files I was able to compile several modules form your Esp-Homekit-demo git. But I can t figure out how to compile this one? could you help? I tried several options but they all end in errors.

I read in the closed Issue section that you are still working on this project? It would be great If it could work like a door bell (e.g. Ring / https://eu.ring.com), but that aside for now.

I have my hardware ready, I only need to know how to compile this code?
Thank you in advance.

Korcsog, 2019-05-20:
@AchimPieters Great idea! I'm using esp-homekit-demo since nearly the beginning, but i also can't compile this one! ESP-IDF also has has bugs in the latest release! Tons of issues opened since the last 1-2 weeks so my question is:
@maximkulkin which release of ESP-IDF did you use? Because the latest might not work well due to bugs..

maximkulkin, 2019-05-20:
@AchimPieters there is a separate instruction on how to compile ESP32 projects in esp-homekit-demo repository wiki. For those you do not need docker, just download precompiled toolchain for your platform from Espressif site, git clone esp-idf repo and set two environment variables 

@Korcsog I do not use these accessories in production yet, I mostly do dev: change something, flash and do a quick test. So I do not experience any problems. Anyways, if esp-idf has problems, we’ll have to wait until they fix them or bisect into esp-idf history to find version that does work

Korcsog, 2019-05-20:
@maximkulkin  Actually I've updated the esp-idf and the camera works (M5Stack Fisheye version)! There was a couple bugs in the latest release, but they fixed these and everything compiles! 
The snapshot works with the camera, but streaming is not fully working yet! Just says 'Loading..." but  nothing really happens..

AchimPieters, 2019-05-21:
@Korcsog So I have installed the esp-idf repo kind off. Thanks for that! this is a whole other :cookie:! 

@maximkulkin I did it I have compiled the code. Now how to get it upon my ESP32. there are multiple .bine files generated.

This is really something I have to dive into. This is really different than the ESP8266 steps.
[image]

But eventually I will get there, I hope...

AchimPieters, 2019-05-21:
@maximkulkin 

`make flash` 

:wink:

maximkulkin, 2019-05-21:
@AchimPieters flashing ESP32 is not much different from ESP8266.. You can just do `make flash` (after you've set ESPPORT environment variable). ESP-IDF has a nice "monitor" application that allows you to flash without exiting monitor: first start it with `make monitor` then inside you can press "Ctrl+T Ctrl+A" and it will do flashing and return back to monitor. You can do a bunch of other stuff (e.g. "Ctrl+T Ctrl+R" to reset ESP). To exit it press "Ctrl+T Ctrl+]"

maximkulkin, 2019-05-21:
Also, you can do `make help` to see list of common targets

Korcsog, 2019-05-22:
@AchimPieters  Also you can do `make flash monitor `. After flashing, `monitor` starts  immediately.

@maximkulkin  Is there any settings for change the resolution for better quality? JPG quality isn't make any changes. I've also set everything else to VGA (SVGA is too big, got some errors) and nothing happens. For the snapshot image 320x240 is too low quality for me, also the image has a lot of artifacts. Until streaming not working, better quality snapshots is a great solution!

AchimPieters, 2019-05-22:
@maximkulkin & @Korcsog Thank you for your quick reply!

Yesterday I flashed My ESP32 but it gave some errors, I didn't had time to debug, becuase my little princess needed some daddy time :relaxed:. Going to try again after work.

I'v tried to made a bash file to install the ESP toolchain, so I will be easier  for others that struggle with the installation:

```
#!/usr/bin/env bash

#Install Prerequisites
sudo easy_install pip

# Setup Toolchain
mkdir -p ~/esp32
cd ~/esp32

curl -O https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
tar -xzf xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz

#update PATH environment variable
echo "export PATH=$HOME/esp32/xtensa-esp32-elf/bin:$PATH" >> /etc/paths

#Get ESP-IDF
cd ~/esp32
git clone -b v3.2 --recursive https://github.com/espressif/esp-idf.git

#update PATH environment variable
echo "export IDF_PATH=~/esp/esp-idf" >> /etc/paths

#Install the Required Python Packages
python -m pip install --user -r $IDF_PATH/requirements.txt

#Start a Test Project

cd ~/esp32
cp -r $IDF_PATH/examples/get-started/hello_world .

cd ~/esp32/hello_world
make menuconfig
```

AchimPieters, 2019-05-22:
unfortunately  I get this error when I flash the EPS32 en do a make monitor:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6188
load:0x40078000,len:10180
load:0x40080400,len:6640
entry 0x40080760
I (29) boot: ESP-IDF v3.2 2nd stage bootloader
I (29) boot: compile time 19:28:40
I (29) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 001f0000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xa1f04 (663300) map
I (324) esp_image: segment 1: paddr=0x000b1f2c vaddr=0x3ffb0000 size=0x039fc ( 14844) load
I (330) esp_image: segment 2: paddr=0x000b5930 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/achimpieters/esp32/esp-idf/components/freertos/xtensa_vectors.S:1779

I (331) esp_image: segment 3: paddr=0x000b5d38 vaddr=0x40080400 size=0x0a2d8 ( 41688) load
I (357) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0xfa20c (1024524) map
0x400d0018: _stext at ??:?

I (716) esp_image: segment 5: paddr=0x001ba22c vaddr=0x4008a6d8 size=0x07c4c ( 31820) load
0x4008a6d8: set_most_pwr_reg at ??:?

I (741) boot: Loaded app from partition at offset 0x10000
I (741) boot: Disabling RNG early entropy source...
I (741) cpu_start: Pro cpu up.
I (745) cpu_start: Starting app cpu, entry point is 0x40081054
0x40081054: call_start_cpu1 at /Users/achimpieters/esp32/esp-idf/components/esp32/cpu_start.c:246

I (737) cpu_start: App cpu up.
I (755) heap_init: Initializing. RAM available for dynamic allocation:
I (762) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (768) heap_init: At 3FFBB770 len 00024890 (146 KiB): DRAM
I (775) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (781) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (787) heap_init: At 40092324 len 0000DCDC (55 KiB): IRAM
I (794) cpu_start: Pro cpu start user code

Brownout detector was triggered
```

AchimPieters, 2019-05-26:
I found that, Brownout detector was triggered has something to do with the power. So I'am first going to do some more background debugging. Regarding the compiling part, that's solved! Thank you for that!

closed 2019-05-26 by AchimPieters

alangoveia, 2019-11-11:
Hi @AchimPieters,
I' am getting the same log above. Did you figured it out?
Thanks in advance

AchimPieters, 2019-11-11:
@alangoveia Yes, use a stable power supply and add a capacitor.

AchimPieters, 2019-11-11:
It’s often related to one of the following issues:

    Poor quality USB cable;
    USB cable is too long;
    Board with some defect (bad solder joints);
    Bad computer USB port;
    Or not enough power provided by the computer USB port.

alangoveia, 2019-11-14:
thanks @AchimPieters ,
I changed the power supply and it worked.