maximkulkin/esp32-homekit-camera #16: esp32 camera model

lizzus, 2019-05-29:
Hi there
can you suggest a camera model compatible?

This one is ok? https://amzn.to/2EIMwno

AchimPieters, 2019-05-29:
@lizzus 

That's the same I have, It should work just fine.

lizzus, 2019-05-29:
Did you test? if work I'll buy!
Is a lot of month I wait this moment!!! Thanks @maximkulkin

AchimPieters, 2019-05-29:
Yes I have tested it. Keep in mind that this project of @maximkulkin is still in progress.

lizzus, 2019-05-29:
many thanks @AchimPieters. I'd like to build a version with OTA. Do you think is possible?

AchimPieters, 2019-05-29:
No not at this point I think, this is a question Maxim can better answer.

lizzus, 2019-05-29:
ok, thanks :-)

maximkulkin, 2019-06-05:
Well, I'm not that familiar with OTA. Given that firmware size for camera is pretty large, I'm not sure it will work.

pwallner, 2019-06-06:
I will try this, much cheaper
https://s.click.aliexpress.com/e/b5h6hpb2

lizzus, 2019-06-06:
@maximkulkin yes is large, but esp32 have 4mb of memory. Right?

maximkulkin, 2019-06-06:
@pwallner I believe, that is the same module, just without Amazon added cost.

@lizzus It's not that simple. Last time I checked OTA had two partitions with two copies of firmware: the current one and the updated one, OTA code swaps which one is current after update. So, said that currently firmware is large enough to not being able to fit twice onto flash.

pwallner, 2019-06-12:
@maximkulkin I saw that it doesn't use the AI thinker, so I also found this: https://de.aliexpress.com/item/32947577882.html?spm=a2g0s.9042311.0.0.17d14c4dwTPkRF

Bit more expensive, but much more cheaper than amazon. I orded at both shops. I will report which one is working.

maximkulkin, 2019-06-12:
@pwallner I am very aware. I have that one available, but for most of my development and testing I'm using [ESP-EYE](https://www.espressif.com/en/products/hardware/esp-eye/overview). The thing I do not like about modules you've mentioned is that they do not come with USB-to-SERIAL chip and thus you have to use external USB-to-SERIAL adapter.

pwallner, 2019-06-12:
yes that's right, so I use this one from [amazon](https://www.amazon.de/gp/product/B01CYBHM26/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1). I already have this usb-to-serial adapter, so the cheap version of esp32 from aliexpress are ok for me (I hope so).

pwallner, 2019-06-25:
I can confirm that the first mentioned module is working like the AI Thinker Chip.

beckmx, 2019-07-15:
hello @pwallner I was testing also the same module you have in [THIS URL](https://www.ebay.com/itm/ESP32-CAM-ESP32-5V-WIFI-Bluetooth-Development-Board-OV2640-Camera-Module/362593241691?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3D4f7c90b2cdee4948b0068c559023b6b3%26pid%3D100675%26rk%3D4%26rkt%3D15%26mehot%3Dpp%26sd%3D163757400118%26itm%3D362593241691%26pg%3D2481888&_trksid=p2481888.c100675.m4236&_trkparms=pageci%3Acf4af8ce-a72b-11e9-8a34-74dbd18011db%7Cparentrq%3Af6d4106b16b0acc7c67b7944ff73f1b6%7Ciid%3A1) is the same ESP32cam, but some shops say PSRAM 4mb external. I was wondering if you guys know if there is a RTOS method to know the total RAM -installed-. Also @lizzus and @AchimPieters are you getting video live out of your modules?

maximkulkin, 2019-07-15:
@beckmx FreeRTOS is a universal platform, it has no ability to detect hardware capabilities. This is a task for "port" - binding of FreeRTOS to particular hardware. I guess, ESP32 has a way to query size of SPI RAM, because you do not need to specify RAM size, it detects it automatically. You can see all regions configured as heap memory in output during boot.

But keep in mind: ESP32 architecture is rather limited in terms of how much memory you can normally address. The memory area that can be mapped to external RAM is only 4MB large. To access more you need to use bank switching techniques (you map different external RAM areas to the same memory area by using special SPI command). Thus the code should be aware of that fact, keep track of what area it is looking at and optimize access so that it makes less expensive bank switches (expensive because of memory cache invalidation).

beckmx, 2019-07-15:
I get it now more in context, I need to keep your pace :), I am happy with getting the screenshots meanwhile (even with the external psram in makemenu), thanx for the clarifications =D

beckmx, 2019-07-16:
I noticed that in the logs in the beggining it shows the external memory added:
`·[0;32mI (1780) heap_init: Initializing. RAM available for dynamic allocation:·[0m
·[0;32mI (1780) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM·[0m
·[0;32mI (1786) heap_init: At 3FFBBC40 len 000243C0 (144 KiB): DRAM·[0m
·[0;32mI (1793) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM·[0m
·[0;32mI (1799) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM·[0m
·[0;32mI (1806) heap_init: At 40096B18 len 000094E8 (37 KiB): IRAM·[0m
·[0;32mI (1812) cpu_start: Pro cpu start user code·[0m
·[0;32mI (1817) spiram: Adding pool of 4096K of external SPI memory to heap allocator·[0m`

closed 2019-09-16 by maximkulkin