maximkulkin/esp32-homekit-camera #17: malloc error

GermanSheepDog, 2019-05-30:


GermanSheepDog, 2019-05-30:
Hello Maxim,
I did this:
1. git clone --recursive https://github.com/maximkulkin/esp32-homekit-camera.git
2. make menuconfig
    - Set serial flasher config to correct USB
    - Flash size to 4MB
    - Set ESP32 HomeKit Camera to my model ESP32_WROVER
    - Set Partition Table to partitions.csv

If I started esp32-homekit-camera it connected to LAN and show the image. If I touch the
picture to get the stream I got an malloc error. Is this at this time of you development correct?

```
I (34003) esp32_camera: Resource payload: {"image-width":640,"image-height":480,"resource-type":"image"}
>>> HomeKit: [Client 57] Update Characteristics
>>> HomeKit: [Client 58] Get Characteristics
>>> HomeKit: [Client 58] Update Characteristics
>>> HomeKit: [Client 58] Get Characteristics
I (39553) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 58] Update Characteristics
I (39633) esp32_camera: Initializing encoder
I (39633) esp32_camera: Total free memory: 131696
I (39633) esp32_camera: Largest free block: 113804
x264 [info]: using cpu capabilities: none!
x264 [error]: malloc of size 4098 failed
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
```

maximkulkin, 2019-06-05:
You need module with external SPI RAM and you need to enable it in configuration: "Components -> ESP32-specific -> Support for external, SPI-connected RAM"

GermanSheepDog, 2019-06-07:
Thank you.
Ok, I changed in configuration: "Component config -> ESP32-specific -> Support for external, SPI-connected RAM" to "*" and I didn't change any other configuration.
I got an error:
``` 
Generating esp32.common.ld
LD build/esp32-homekit-camera.elf
/Users/myname/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/myname/esp/esp32-homekit-camera/build/esp32-homekit-camera.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/Users/myname/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/Users/myname/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 25192 bytes
collect2: error: ld returned 1 exit status
make: *** [/Users/myname/esp/esp32-homekit-camera/build/esp32-homekit-camera.elf] Error 1
Pg1:esp32-homekit-camera myname$ 
```
I used a TTGO T-Kamera ESP32 WROVER & PSRAM Camera with PSRAM: 8MBytes. 

I suspect that another parameter needs to be changed.

maximkulkin, 2019-06-07:
You need to update eps-idf SDK to a more recent version

GermanSheepDog, 2019-06-08:
Thank you for your great help, I can compile and download the software. Now I used SPI Flash Size with 4MB.

At homekit I saw images and this worked fine. If I like to get videos I got an ' Core  1 panic'ed (LoadProhibited). Exception was unhandled' error:
```
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Resource
I (27501) esp32_camera: Resource payload: {"image-width":640,"image-height":480,"resource-type":"image"}
>>> HomeKit: [Client 55] Resource
I (71231) esp32_camera: Resource payload: {"image-width":640,"image-height":480,"resource-type":"image"}
>>> HomeKit: [Client 55] Get Characteristics
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Get Characteristics
I (77451) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 55] Update Characteristics
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008ee8f  PS      : 0x00060630  A0      : 0x80084d04  A1      : 0x3ffeba10  
0x4008ee8f: is_free at /Users/myname/esp/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /Users/myname/esp/esp-idf/components/heap/multi_heap.c:432

A2      : 0x3f800000  A3      : 0x000e1000  A4      : 0x00000000  A5      : 0x3ffdec30  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00000000  A9      : 0xffffffff  
A10     : 0x3f800010  A11     : 0x3ffbcd14  A12     : 0x3f800014  A13     : 0x00000000  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x4008fd80  LEND    : 0x4008fd8b  LCOUNT  : 0x00000000  
0x4008fd80: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x4008fd8b: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

ELF file SHA256: 15af80b002dee1e2696864daa7ea98116bd7f210e96a49d49b313214726fa5a4

Backtrace: 0x4008ee8f:0x3ffeba10 0x40084d01:0x3ffeba30 0x40084d53:0x3ffeba50 0x4008b34d:0x3ffeba70 0x400d7cd5:0x3ffeba90 0x400915f5:0x3ffec2e0
0x4008ee8f: is_free at /Users/myname/esp/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /Users/myname/esp/esp-idf/components/heap/multi_heap.c:432

0x40084d01: heap_caps_malloc at /Users/myname/esp/esp-idf/components/heap/heap_caps.c:354

0x40084d53: heap_caps_malloc_default at /Users/myname/esp/esp-idf/components/heap/heap_caps.c:354

0x4008b34d: malloc at /Users/myname/esp/esp-idf/components/newlib/heap.c:32

0x400d7cd5: camera_stream_task at /Users/myname/esp/esp32-homekit-camera/main/app_main.c:589

0x400915f5: vPortTaskWrapper at /Users/myname/esp/esp-idf/components/freertos/port.c:435

Rebooting...
I (12) boot: ESP-IDF v4.0-dev-728-g826ff7186 2nd stage bootloader
I (12) boot: compile time 10:13:57
I (12) boot: Enabling RNG early entropy source...
I (17) boot: SPI Speed      : 40MHz
I (21) boot: SPI Mode       : DIO
I (25) boot: SPI Flash Size : 4MB
I (29) boot: Partition Table:
I (33) boot: ## Label            Usage          Type ST Offset   Length
I (40) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (47) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (55) boot:  2 factory          factory app      00 00 00010000 001f0000
I (62) boot: End of partition table
I (67) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xaf5b0 (718
```

maximkulkin, 2019-07-10:
DId you enable SPI RAM in menuconfig?

GermanSheepDog, 2019-07-11:
Yes, I think I did it. Here are some lines from sdkconfig:
```
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_BOOT_INIT=y
```

lockpicker, 2019-07-14:
I'm getting the same error, logs below: 
`>>> HomeKit: [Client 55] Update Characteristics
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008ba82  PS      : 0x00060333  A0      : 0x8008dbad  A1      : 0x3ffeb9c0  
0x4008ba82: uxPortCompareSet at /home/lockpicker/esp/esp-idf/components/freertos/tasks.c:3537
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/lockpicker/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/lockpicker/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/lockpicker/esp/esp-idf/components/freertos/tasks.c:4231
 
A2      : 0xffff0000  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060320  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffeb9b0  
A10     : 0x3ffe0fc0  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xffff0000  LBEG    : 0x40095d3c  LEND    : 0x40095d47  LCOUNT  : 0x00000000  
0x40095d3c: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142
 
0x40095d47: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152
 
 
Backtrace: 0x4008ba82:0x3ffeb9c0 0x4008dbaa:0x3ffeb9f0 0x40085270:0x3ffeba10 0x400852c1:0x3ffeba30 0x400857c5:0x3ffeba50 0x4000beaf:0x3ffeba70 0x400d7c8c:0x3ffeba90 0x4008ad59:0x3ffec2e0
0x4008ba82: uxPortCompareSet at /home/lockpicker/esp/esp-idf/components/freertos/tasks.c:3537
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/lockpicker/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/lockpicker/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/lockpicker/esp/esp-idf/components/freertos/tasks.c:4231
 
0x4008dbaa: multi_heap_internal_lock at /home/lockpicker/esp/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /home/lockpicker/esp/esp-idf/components/heap/multi_heap.c:416
 
0x40085270: heap_caps_malloc at /home/lockpicker/esp/esp-idf/components/heap/heap_caps.c:354
 
0x400852c1: heap_caps_malloc_default at /home/lockpicker/esp/esp-idf/components/heap/heap_caps.c:354
 
0x400857c5: _malloc_r at /home/lockpicker/esp/esp-idf/components/newlib/syscalls.c:37
 
0x400d7c8c: camera_stream_task at /home/lockpicker/Developer/esp32-homekit-camera/main/app_main.c:589

0x4008ad59: vPortTaskWrapper at /home/lockpicker/esp/esp-idf/components/freertos/port.c:435

Rebooting...`

maximkulkin, 2019-07-14:
Make sure you follow configuration instructions in README that I added recently.

GermanSheepDog, 2019-07-15:
@maximkulkin: I started new from scratch:
1. I deleted directory 'esp32-homekit-camera'
2. Download the whole software: 'git clone --recursive https://github.com/maximkulkin/esp32-homekit-camera.git'
3. Partition Table = Custom partition table CSV, Custom partition CSV file = partitions.csv
4. Component config: ESP32-specific
    - Support for external, SPI-connected RAM = ok
    - SPI RAM config: Initialize SPI RAM when booting the ESP32 = ok
    - SPI RAM access method = Make RAM allocatable using malloc() as well = ok
5. Camera configuration: OV2640 Support = ok
6. ESP32 HomeKit Camera: STA
7. Camera Pins = custom camera pinout:
#define CAMERA_PIN_PWDN   26
#define CAMERA_PIN_RESET  -1
#define CAMERA_PIN_SIOD   13
#define CAMERA_PIN_SIOC   12
#define CAMERA_PIN_XCLK   32
#define CAMERA_PIN_D0      5
#define CAMERA_PIN_D1     14
#define CAMERA_PIN_D2      4
#define CAMERA_PIN_D3     15
#define CAMERA_PIN_D4     18
#define CAMERA_PIN_D5     23
#define CAMERA_PIN_D6     36
#define CAMERA_PIN_D7     39
#define CAMERA_PIN_VSYNC  27
#define CAMERA_PIN_HREF   25
#define CAMERA_PIN_PCLK   19
This is a  WROVER_Kit from TTgo (exactly: TTGO T-Kamera ESP32 WROVER & PSRAM Kamera Modul ESP32-WROVER-B OV2640 Kamera Modul 0,96 OLED).

Results:
a.) Program started, Wifi worked, homekit pairing worked, image loading every 10 seconds worked very well!
b.) If I touch the picture to get the stream I got the 'Guru Meditation Error: Core  1 panic'ed'
```
I (414330) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 56] Update Characteristics
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008f34b  PS      : 0x00060330  A0      : 0x80084dc8  A1      : 0x3ffeba10  
0x4008f34b: is_free at /Users/myname/esp/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /Users/myname/esp/esp-idf/components/heap/multi_heap.c:432

A2      : 0x3f800000  A3      : 0x000e1000  A4      : 0x00000000  A5      : 0x3ffc36d8  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0xffffffff  
A10     : 0x3f800010  A11     : 0x3ff96354  A12     : 0x3f800014  A13     : 0x00000000  
A14     : 0xffffffff  A15     : 0x00000057  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x4008e7e4  LEND    : 0x4008e7ef  LCOUNT  : 0x00000000  
0x4008e7e4: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x4008e7ef: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

ELF file SHA256: 681ccb6492e90ad4b7bddfe5fa454ba0111d4055c7330e8530024ec1748525c5

Backtrace: 0x4008f348:0x3ffeba10 0x40084dc5:0x3ffeba30 0x40084e17:0x3ffeba50 0x400931fd:0x3ffeba70 0x400d7d2d:0x3ffeba90 0x40093499:0x3ffec2e0
0x4008f348: multi_heap_assert at /Users/myname/esp/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /Users/myname/esp/esp-idf/components/heap/multi_heap.c:431

0x40084dc5: heap_caps_malloc at /Users/myname/esp/esp-idf/components/heap/heap_caps.c:354

0x40084e17: heap_caps_malloc_default at /Users/myname/esp/esp-idf/components/heap/heap_caps.c:354

0x400931fd: malloc at /Users/myname/esp/esp-idf/components/newlib/heap.c:32

0x400d7d2d: camera_stream_task at /Users/myname/esp/esp32-homekit-camera/main/app_main.c:589

0x40093499: vPortTaskWrapper at /Users/myname/esp/esp-idf/components/freertos/port.c:435

Rebooting...
```
From my point of view, I suspect no problem with type of camera or with the pins. I suspect a problem with heap. Is there any way to get more debugging informations?

@lockpicker: What kind of camera did you use?

AchimPieters, 2019-07-17:
Streaming does not work (Yet). See  #11 and #22

GermanSheepDog, 2019-07-19:
@AchimPieters 
Thanks a lot for this information.
I saw that Issue #11 is closed and suspects that streaming is now working. I did not found Issue #22. 
So, now I'm waiting with Maxim...

closed 2019-09-16 by maximkulkin

Kirankumar-xenlabs, 2019-10-11:
@GermanSheepDog I am also getting same problem