maximkulkin/esp32-homekit-camera #19: Compile error

pwallner, 2019-06-20:
Debian 9.8 x64

```
mcpat@debian:~/esp/esp32-homekit-camera$ make all
Toolchain path: /home/mcpat/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /home/mcpat/esp/esp-idf/requirements.txt are satisfied.

App "esp32-homekit-camera" version: ab7d9e5
CC build/homekit/src/storage.o
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_init':
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:56:38: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
         strncpy(magic, magic1, sizeof(magic1));
                                      ^
In file included from /home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:6:
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_can_add_pairing':
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/port.h:20:85: warning: value computed is not used [-Wunused-value]
 #define spiflash_read(addr, buffer, size) (spi_flash_read((addr), (buffer), (size)) == ESP_OK)
                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:149:9: note: in expansion of macro 'spiflash_read'
         spiflash_read(PAIRINGS_ADDR + sizeof(data)*i, (byte *)&data, sizeof(data));
         ^~~~~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'find_empty_block':
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/port.h:20:85: warning: value computed is not used [-Wunused-value]
 #define spiflash_read(addr, buffer, size) (spi_flash_read((addr), (buffer), (size)) == ESP_OK)
                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:202:9: note: in expansion of macro 'spiflash_read'
         spiflash_read(PAIRINGS_ADDR + sizeof(data)*i, data, sizeof(data));
         ^~~~~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_add_pairing':
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:233:39: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
     strncpy(data.magic, magic1, sizeof(magic1));
                                       ^
In file included from /home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:6:
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_update_pairing':
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/port.h:20:85: warning: value computed is not used [-Wunused-value]
 #define spiflash_read(addr, buffer, size) (spi_flash_read((addr), (buffer), (size)) == ESP_OK)
                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:257:9: note: in expansion of macro 'spiflash_read'
         spiflash_read(PAIRINGS_ADDR + sizeof(data)*i, (byte *)&data, sizeof(data));
         ^~~~~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_remove_pairing':
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/port.h:20:85: warning: value computed is not used [-Wunused-value]
 #define spiflash_read(addr, buffer, size) (spi_flash_read((addr), (buffer), (size)) == ESP_OK)
                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:294:9: note: in expansion of macro 'spiflash_read'
         spiflash_read(PAIRINGS_ADDR + sizeof(data)*i, (byte *)&data, sizeof(data));
         ^~~~~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_find_pairing':
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/port.h:20:85: warning: value computed is not used [-Wunused-value]
 #define spiflash_read(addr, buffer, size) (spi_flash_read((addr), (buffer), (size)) == ESP_OK)
                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:315:9: note: in expansion of macro 'spiflash_read'
         spiflash_read(PAIRINGS_ADDR + sizeof(data)*i, (byte *)&data, sizeof(data));
         ^~~~~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_next_pairing':
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/port.h:20:85: warning: value computed is not used [-Wunused-value]
 #define spiflash_read(addr, buffer, size) (spi_flash_read((addr), (buffer), (size)) == ESP_OK)
                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/mcpat/esp/esp32-homekit-camera/components/homekit/src/storage.c:363:9: note: in expansion of macro 'spiflash_read'
         spiflash_read(PAIRINGS_ADDR + sizeof(data)*id, (byte *)&data, sizeof(data));
         ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [src/storage.o] Error 1
make: *** [component-homekit-build] Error 2
```

maximkulkin, 2019-06-24:
Looks like it's a new check in compiler added recently, should be fine to disable it with `-Wno-error=sizeof-pointer-memaccess`

pwallner, 2019-06-25:
I will try with `-Wno-error=sizeof-pointer-memaccess`, do you know how to set this?

Now I used Mingw32 (acc. [Docs Espressif](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html) but with warnings during compilation:
```
 WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a5
Expected to see version: esp32-2019r1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
```

maximkulkin, 2019-06-25:
No, I do not do Windows support, sorry

cranor, 2019-07-05:
The secure way :

line 55 : strncpy(magic, magic1, sizeof(magic1)<sizeof(magic)?sizeof(magic1):sizeof(magic));
line 233 : strncpy(data.magic, magic1, sizeof(magic1)<sizeof(data.magic)?sizeof(magic1):sizeof(data.magic));

maximkulkin, 2019-07-05:
Nope, not doing that. If you look at the code, `magic` IS defined to have size of magic1.

maximkulkin, 2019-07-05:
I have already pushed a fix, should not complain now.

closed 2019-07-05 by maximkulkin

cranor, 2019-07-06:
libs are also affected by this compiler check.

Not the safest but simplest solution:

make menuconfig
 - compiler options
   - Disable new warnings introduced in GCC 6 - 8