maximkulkin/esp32-homekit-camera #22: Deprioritize -> Prioritized

AchimPieters, 2019-06-28:
@maximkulkin 

I read in #11 that you deprioritized the steaming issue.
Are you willing to prioritized this again? It would be really appreciated!  :pray:

maximkulkin, 2019-07-17:
I feel I'm pretty close, but I periodically run into hardware problems which slow me down a lot. Last one - my ESP-EYE module sometimes start to have issues with getting data from sensor. It just gets frozen. So annoying. I replaced sensor from another module and it seemed to work, but then the problem returned. Seems like some problem with connector/cable. I decided to replace the whole module. So, now I'm waiting for it.

AchimPieters, 2019-07-17:
@maximkulkin That's great news, I really hope you get it working, unfortunately it's a little above my pay grade, so I'm relying on your expertise. But if I can help or support you in any way please let me know!

beckmx, 2019-07-25:
@maximkulkin I can send u some kudo$ to ur paypal, I use ur work in my house a lot so, I woulnt mind to cooperate

GermanSheepDog, 2019-09-17:
@maximkulkin Is there any news from streaming? Today I downloaded the whole esp32-homekit-camera-software and tested it with CONFIG_CAMERA_MODEL_TTGO_V1_7. I got the old malloc error:

```
0x4008f34b: is_free at /Users/username/esp/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /Users/username/esp/esp-idf/components/heap/multi_heap.c:432
```

beckmx, 2019-09-17:
@GermanSheepDog and the rest, I was testing outside the homekit tests to actually test the streaming capabilities, I see now that for streaming it does require a lot more RAM, and with thecurrent setup will be very difficult due to the rest of the tasks the processor needs to handle, specially for encrypting the data, even without the homekit library, streaming fails inconsistently when adding a couple of things like mqtt

GermanSheepDog, 2019-09-18:
@beckmx Thank you for the interesting information. I keep my fingers crossed.

I have also added a variety code to esp_homekit and got big problems with the heap. Now I rewrote all my l libraries and it worked fine.

maximkulkin, 2019-10-11:
Ok, I have something working. Unfortunately, it seems like bug in WiFi stack often makes camera unresponsive. I hope that it will be fixed in ESP-IDF. You can recognize this problem when you see this output in console:
```
W (12211) wifi: alloc eb len=24 type=3 fail, heap:3552232

W (12211) wifi: m f null
```
Shortly after that you get network error and it seems that packets are not flowing. 

Meantime I'm experimenting with code hoping to find a workaround.

It turned out that memory was not the biggest issue. The biggest issue is not being able to encode frames at decent rate. Thus I had to downscale encoded image 8 times (from 640x480 down to 80x60) to get something at least somewhat usable.

Also, very rarely JPEG decoding freezes. Not sure why. Need more stable repro.

But anyways, congratulations everybody with a milestone!

closed 2019-10-11 by maximkulkin