maximkulkin/esp32-homekit-camera #23: Compilation homekit-build

beckmx, 2019-07-11:
Hello Max,

I downloaded the code, setup my environment and got stuck in this issue:
`I make: *** No rule to make target 'component-homekit-build', needed by 'component-main-build'.  Stop.`

I will look in your code base to see if something is missing but I notice it might be missing something.

Cheers!

beckmx, 2019-07-11:
nvm, recursive clone

closed 2019-07-11 by beckmx

ptath, 2020-01-23:
@beckmx same error, what does it means "nvm, recursive clone" and how to beat this? ) thanks

maximkulkin, 2020-01-23:
@ptath 
After you've checked it out, make sure you update all submodules
```
git submodule update --init --recursive
```
and if you've built it before, it makes sense to do `make clean` in your project.

beckmx, 2020-01-23:
@ptath one simple way to clone the repo is just to do:
`git clone http://xxxxxxx --recursive`

ptath, 2020-01-23:
@maximkulkin @beckmx thank you a lot, already figured it out. Camera added, but shows only snapshots, no video. Strange things in serial logs:

```
x264 [info]: using cpu capabilities: none!
x264 [error]: malloc of size 4098 failed
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
```

Full log:
```
I (15699) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: Got new client connection: 57
>>> HomeKit: [Client 57] Pair Verify Step 1/2
>>> HomeKit: [Client 57] Pair Verify Step 2/2
>>> HomeKit: [Client 57] Found pairing with FF771E6A-F3EF-4D34-8C51-6B4AB69D6102
>>> HomeKit: [Client 57] Verification successful, secure session established
>>> HomeKit: [Client 57] Get Accessories
I (25839) esp32_camera: Creating setup endpoints response
>>> HomeKit: Got new client connection: 58
>>> HomeKit: [Client 58] Pair Verify Step 1/2
>>> HomeKit: [Client 58] Pair Verify Step 2/2
>>> HomeKit: [Client 58] Found pairing with FF771E6A-F3EF-4D34-8C51-6B4AB69D6102
>>> HomeKit: [Client 58] Verification successful, secure session established
>>> HomeKit: [Client 58] Get Accessories
I (27179) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Resource
I (30749) esp32_camera: Resource payload: {"image-width":640,"image-height":480,"resource-type":"image"}
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Get Characteristics
I (35139) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 56] Update Characteristics
I (36029) esp32_camera: Initializing encoder
I (36029) esp32_camera: Total free memory: 132608
I (36029) esp32_camera: Largest free block: 113804
x264 [info]: using cpu capabilities: none!
x264 [error]: malloc of size 4098 failed
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4018a772  PS      : 0x00060e30  A0      : 0x801888a4  A1      : 0x3ffdeaf0  
0x4018a772: jpeg_idct_islow at /Users/ptath/esp32/esp32-homekit-camera/components/libjpeg/jpeg-6b/jidctint.c:360

A2      : 0x00000000  A3      : 0x3ffe461c  A4      : 0xfffd39d0  A5      : 0x00000048  
A6      : 0x00020000  A7      : 0x00000000  A8      : 0x3ffdeaf0  A9      : 0xfffe6db0  
A10     : 0xffff72c0  A11     : 0x00003a80  A12     : 0xfffda5f0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x0001ccd0  SAR     : 0x00000014  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 7de128035de46508ab77d7c87f17dbdf4aa859c5130f0d0df33ecbe36ea4339f

Backtrace: 0x4018a772:0x3ffdeaf0 0x401888a1:0x3ffdec40 0x401ca739:0x3ffdec90 0x400d4090:0x3ffdecb0 0x4008c339:0x3ffdf500
0x4018a772: jpeg_idct_islow at /Users/ptath/esp32/esp32-homekit-camera/components/libjpeg/jpeg-6b/jidctint.c:360

0x401888a1: decompress_onepass at /Users/ptath/esp32/esp32-homekit-camera/components/libjpeg/jpeg-6b/jdcoefct.c:197 (discriminator 3)

0x401ca739: jpeg_read_raw_data at /Users/ptath/esp32/esp32-homekit-camera/components/libjpeg/jpeg-6b/jdapistd.c:210

0x400d4090: camera_stream_task at /Users/ptath/esp32/esp32-homekit-camera/main/app_main.c:424

0x4008c339: vPortTaskWrapper at /Users/ptath/esp32/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

```

maximkulkin, 2020-01-23:
@ptath you seem to have a module without additional SPI RAM (Notice before starting streaming you have only 130Kb of RAM: `Total free memory: 132608`). Such modules do not have enough memory to run live streaming.

ptath, 2020-01-24:
Totally my fault, wrong config (missed SPI flash address for storing HomeKit data = 0x3A0000). 

Now camera works a few seconds (10-20) and stops. Black screen, no errors in HK or serial, works fine after reloading stream.

Anyway, this is a great progress, thanks )