maximkulkin/esp32-homekit-camera #24: End of partition table on esp32cam

beckmx, 2019-07-13:
Hello guys,

So I compiled my code with the options:
```
 AP or STA (SoftAP)  --->                                                                                          │ │
  │ │                                               (Unknown2) WiFi SSID                                                                                                  │ │
  │ │                                               (12345678) WiFi Password                                                                                              │ │
  │ │                                               (3) Max STA conn                                                                                                      │ │
  │ │                                               (20000000) XCLK Frequency                                                                                             │ │
  │ │                                                   Camera Pins  --->
```

Under the menu camera pins i selected:
`ESP32-CAM by AI-Thinker`
Which [is this one I believe ](https://de.aliexpress.com/item/32953149742.html?spm=a2g0x.search0204.3.57.bd4f69f8BD7fVM&ws_ab_test=searchweb0_0%2Csearchweb201602_4_10065_10068_319_10546_10059_10884_317_10548_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_537_536%2Csearchweb201603_52%2CppcSwitch_0&algo_expid=d1617d7f-63e1-4baf-984c-dfa29a19ea13-9&algo_pvid=d1617d7f-63e1-4baf-984c-dfa29a19ea13) The code looks to be compiled and then I upload using these configs:

`esptool.py --chip esp32 --port /dev/tty.usbserial-A800K5UT --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 build/bootloader/bootloader.bin 0x10000 build/esp32-homekit-camera.bin 0x8000 build/partitions_singleapp.bin`

I see the code uploaded correctly but if I connect to the terminal with UART I see this wierd message:
`rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6444
load:0x40078000,len:11492
load:0x40080400,len:6664
entry 0x40080764
·[0;32mI (56) boot: ESP-IDF v4.0-dev-1191-g138c941 2nd stage bootloader·[0m
·[0;32mI (56) boot: compile time 04:45:11·[0m
·[0;32mI (57) boot: Enabling RNG early entropy source...·[0m
·[0;32mI (63) boot: SPI Speed      : 40MHz·[0m
·[0;32mI (67) boot: SPI Mode       : DIO·[0m
·[0;32mI (71) boot: SPI Flash Size : 4MB·[0m
·[0;32mI (75) boot: Partition Table:·[0m
·[0;32mI (79) boot: ## Label            Usage          Type ST Offset   Length·[0m
·[0;32mI (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000·[0m
·[0;32mI (94) boot:  1 phy_init         RF data          01 01 0000f000 00001000·[0m
·[0;32mI (101) boot:  2 factory          factory app      00 00 00010000 00100000·[0m
·[0;32mI (109) boot: End of partition table·[0m
·[0;32mI (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xa6e94 (683668) map·[0m
·[0;32mI (361) esp_image: segment 1: paddr=0x000b6ebc vaddr=0x3ffb0000 size=0x03d98 ( 15768) load·[0m
·[0;32mI (368) esp_image: segment 2: paddr=0x000bac5c vaddr=0x40080000 size=0x00400 (  1024) load·[0m
·[0;32mI (368) esp_image: segment 3: paddr=0x000bb064 vaddr=0x40080400 size=0x04fac ( 20396) load·[0m
·[0;32mI (385) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0x101274 (1053300) map·[0m
·¸ets Jun  8 2016 00:22:57`

And keeps repeating over and over, anyone knows if the configs should be different?

maximkulkin, 2019-07-13:
You need to set "Partition Table -> Partition Table -> Custom partition table CSV" in menuconfig. Also, make sure that "Partition Table -> Custom partition CSV file" is set to "partitions.csv" (should be default). I will update README with all required settings.

maximkulkin, 2019-07-13:
Ok, I've updated README, make sure you've done all required settings. Let me know if that won't work.

beckmx, 2019-07-13:
:)) it did work, I just discovered the same issue on the other thread about not streaming video, it does take the pictures though, but according to that, you say an esp32 with external RAM is required right?, what device do you use? I sed one of the cheap esp32cam but I cannot find what devices implement the external ram yet.

Greetings!

maximkulkin, 2019-07-14:
Snapshot should work even without SPI RAM. What does your module do instead?

Modules with SPI RAM can be easily found: [google "esp32 camera spi ram"](https://www.google.com/search?q=esp32%20camera%20spi%20ram)

beckmx, 2019-07-14:
It does the snapshot correctly, everything on place, I added the additional service for the flashlight, but yeah as u said, the streaming needs more ram, what device do u use?

beckmx, 2019-07-14:
how much ram do u think it needs, I found this one https://www.banggood.com/LILYGO-TTGO-T-Camera-ESP32-WROVER-with-PSRAM-Camera-Module-OV2640-Camera-0_96-Inch-OLED-p-1418433.html?ID=566074&cur_warehouse=CN

AchimPieters, 2019-07-14:
Why not use the ons I use? IT has a SD card as extra ram.

Here is a link for Example https://www.google.nl/search?q=esp32+cam+aliexpress&tbm=isch&ved=2ahUKEwjXqKCGrLXjAhWDDFAKHQYxAsgQ2-cCegQIABAC&oq=esp32+cam+ali&gs_l=mobile-gws-wiz-img.1.0.0i24.8615.15741..16739...0.0..0.59.910.17......0....1.........0i30j0j0i19j0i8i30i19.Aj7JHl0UFzw&ei=SZ0rXZfgM4OZwAKG4ojADA&bih=719&biw=414&client=safari&prmd=sivn&hl=nl-nl#imgrc=A99wtwif1GOulM

beckmx, 2019-07-14:
oh really? u can use the SD card as ram? because I have them as well, should I insert a blank SD or specially formated? or just doesnt matter

maximkulkin, 2019-07-15:
Hell no, you can't use SD card as RAM. Also, even if you find module with more than 4MB of RAM (e.g. ESP-EYE has 8MB), transparent usage is only possible for 4MB, for the rest you need to manually switch banks. This might work if you're working with memory directly with your own code, but it's almost impossible to teach third-party libraries about that. So, I personally at this stage (when I want to make third-party H.264 codec work) see no reason to buy modules with more than 4MB of RAM. And, guys, do not rush to buy it: getting more memory is not the only problem there.

beckmx, 2019-07-15:
I will close the issue since it was solved and continue over another thread that has already been opened, thanx @maximkulkin

closed 2019-07-15 by beckmx