maximkulkin/esp32-homekit-camera #26: cmake compilation

beckmx, 2019-08-13:
this is more a question for suggestion, somehow I couldnt compile the code with my standard toolchain so, I setup a fresh brand new toolchain from the latest available and noticed that it won't support Makefile any more (I think) and it now asks for the new CMakelists.txt, do you have a suggestion if there is a tool that could help me recreate the CMake?

:) greetings!

AchimPieters, 2019-08-14:
@beckmx  
Witch version did you install?
Did you update your PATH environment variable?
 
`export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH`

As far I can see you still need to run:

`make menuconfig`

maybe this will help: [Espressif ESP32](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/)

beckmx, 2019-08-16:
nevermind, some libs will also need to be updated and that makes the process harder to maintain

closed 2019-08-16 by beckmx