maximkulkin/esp32-homekit-camera #27: More than 2 on the same network

beckmx, 2019-08-16:
I noticed something weird, when I have up to 2 cameras, they seem to work just fine, but immediately when I connect the third one the others disconnect, I wonder if they disconnect because the same mdns name

maximkulkin, 2019-08-16:
Can't say for sure, but having unique names is generally a good idea.

beckmx, 2019-08-21:
Another doubt, I was on the works to catch this issue and noticed something, I changed the size of the app in the partition table to 0x280000 and in menuconfig expanded the spi flash size from 2M to 4M but I think it doesnt like it:

```
>>> HomeKit: Formatting flash at 0x280000
abort() was called at PC 0x4011c3ca on core 1
```
My new flash size is around 2.5mgs so I guessed that by increasing the "magic" number it could work, sounds to me that area should be free, what do u think?

beckmx, 2019-08-22:
well, looks like the initial 10000 in offset was not considered in my sum,

closed 2019-08-22 by beckmx