maximkulkin/esp32-homekit-camera #28: Compilation error

JKSProds, 2019-08-24:
Mac OS 10.14
GNU Make 3.81

`In function 'copy',
    inlined from 'av_lzo1x_decode' at /Users/jmonteiro/Documents/ESP8266HomeKit/esp32-homekit-camera/components/libswscale/libavutil/lzo.c:197:9:
/Users/jmonteiro/Documents/ESP8266HomeKit/esp32-homekit-camera/components/libswscale/libavutil/lzo.c:105:5: error: 'memcpy' specified size between 4294967292 and 4294967295 exceeds maximum object size 2147483647 [-Werror=stringop-overflow=]
     memcpy(dst, src, cnt);`

beckmx, 2019-08-29:
are you compiling against esp8266?, the compilation is actually very straightforward, did you also use the make menuconfig?

maximkulkin, 2019-09-16:
You need to figure out pins.. Unfortunately I can not do this remotely and without proper equipment. You'd better search for a datasheet or some kind of documentation on the module you're using.

JKSProds, 2019-09-16:
Ok, I've asked the suppler for the pinout. And no more probe error. 

[image]

For anyone o needs it. PINS for DIYMORE Board.

However, every time I open de camera inside the home app the esp32 reboots.

`I (873) cpu_start: App cpu up.
I (1772) spiram: SPI SRAM memory test OK
I (1773) heap_init: Initializing. RAM available for dynamic allocation:
I (1773) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1779) heap_init: At 3FFBBB60 len 000244A0 (145 KiB): DRAM
I (1785) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1792) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1798) heap_init: At 40096268 len 00009D98 (39 KiB): IRAM
I (1805) cpu_start: Pro cpu start user code
I (1809) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (153) cpu_start: Chip Revision: 1
W (154) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
I (157) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (168) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (238) wifi: wifi driver task: 3ffcb724, prio:23, stack:3584, core=0
I (238) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (238) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (288) wifi: wifi firmware version: aeed694
I (288) wifi: config NVS flash: enabled
I (288) wifi: config nano formating: disabled
I (288) wifi: Init dynamic tx buffer num: 32
I (288) wifi: Init data frame dynamic rx buffer num: 32
I (298) wifi: Init management frame dynamic rx buffer num: 32
I (298) wifi: Init management short buffer num: 32
I (308) wifi: Init static tx buffer num: 16
I (308) wifi: Init static rx buffer size: 1600
I (308) wifi: Init static rx buffer num: 10
I (318) wifi: Init dynamic rx buffer num: 32
I (418) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (418) wifi: mode : sta (24:6f:28:0c:9f:60)
STA start
I (1148) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2278) wifi: state: init -> auth (b0)
I (2328) wifi: state: auth -> assoc (0)
I (2388) wifi: state: assoc -> run (10)
I (2718) wifi: connected with Monteiro-WIFI, channel 6, HT20, bssid = 78:11:dc:52:ea:86
I (2718) wifi: pm start, type: 1

I (3728) event: sta ip: 10.0.0.51, mask: 255.255.255.0, gw: 10.0.0.250
WiFI ready
I (3728) esp32_camera: Free heap: 4364344
I (3728) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3738) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3748) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3778) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3858) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3858) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3868) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3878) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3888) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3898) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3908) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3918) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3928) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3938) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3948) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (3958) camera: Allocating 2 frame buffers (18 KB total)
I (3958) camera: Allocating 9 KB frame buffer in OnBoard RAM
I (3968) camera: Allocating 9 KB frame buffer in OnBoard RAM
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: C6:22:A5:D7:B2:EC
[... 72 lines cut ...]
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xaf48c (717964) map
I (344) esp_image: segment 1: paddr=0x000bf4b4 vaddr=0x3ffb0000 size=0x00b5c (  2908) load
I (345) esp_image: segment 2: paddr=0x000c0018 vaddr=0x400d0018 size=0x11a6ec (1156844) map
0x400d0018: _stext at ??:?

I (757) esp_image: segment 3: paddr=0x001da70c vaddr=0x3ffb0b5c size=0x03190 ( 12688) load
I (762) esp_image: segment 4: paddr=0x001dd8a4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/jmonteiro/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (764) esp_image: segment 5: paddr=0x001ddcac vaddr=0x40080400 size=0x15e68 ( 89704) load
I (823) boot: Loaded app from partition at offset 0x10000
I (823) boot: Disabling RNG early entropy source...
I (823) psram: This chip is ESP32-D0WD
I (829) spiram: Found 64MBit SPI RAM device
I (832) spiram: SPI RAM mode: flash 40m sram 40m
I (838) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (845) cpu_start: Pro cpu up.
I (849) cpu_start: Application information:
I (853) cpu_start: Project name:     esp32-homekit-camera
I (860) cpu_start: App version:      f23b543
I (865) cpu_start: Compile time:     Sep 16 2019 22:34:19
I (871) cpu_start: ELF file SHA256:  554606289eddf204...
I (877) cpu_start: ESP-IDF:          v3.3
I (881) cpu_start: Starting app cpu, entry point is 0x40081474
0x40081474: call_start_cpu1 at /Users/jmonteiro/esp/esp-idf/components/esp32/cpu_start.c:270`

maximkulkin, 2019-09-16:
I think by "open camera" you've meant tapping on camera to get into live stream view. That part does not work yet and latest master is not stable in general.

closed 2019-09-16 by maximkulkin