maximkulkin/esp32-homekit-camera #30: Camera issues on latest idf

beckmx, 2019-08-29:
Hello, I downloaded the repo (fresh new twice) it compiles and everything, with 2 different boards, and noticed that the only thing I changed lately is the IDF, just wondering if you noticed it, the problem is present with v4 or after august 13
```
I (3925) esp32_camera: Free heap: 4364644
I (3925) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3935) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (3945) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=3
E (3955) camera_xclk: ledc_timer_config failed, rc=ffffffff
I (3965) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
SCCB_Write [ff]=01 failed
SCCB_Write [12]=80 failed
E (4025) camera: Camera probe failed with error 0x20001
E (4025) esp32_camera: Camera init failed with error 0x20001
```

maximkulkin, 2019-09-05:
Are you sure you’re using correct pin assignment?

beckmx, 2019-09-05:
Yes, actually the only difference was the esp-idf version, and I noticed that they changed the ledc_timer settings

maximkulkin, 2019-09-16:
esp32-camera module repo has an [issue regarding this](https://github.com/espressif/esp32-camera/issues/66). Check it for a workaround while they still working on a fix.

beckmx, 2019-09-17:
I am okay with that info, I will handle from there =D

closed 2019-09-17 by beckmx