maximkulkin/esp32-homekit-camera #33: camera init failed on esp-eye board

gozfree, 2019-09-14:
I have updated esp32-homekit-camera code, and make flash to esp-eye board, but seems camera init failed:
```
...
E (11963) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=3
E (11983) camera_xclk: ledc_timer_config failed, rc=ffffffff 
SCCB_Write [ff]=01 failed
SCCB_Write [12]=80 failed
E (12013) camera: Camera probe failed with error 0x20001
E (12013) esp32_camera: Camera init failed with error 0x20001
...
```
sensor is OV2640, could you help me

maximkulkin, 2019-09-16:
`0x20001` is "[camera not detected](https://github.com/espressif/esp32-camera/blob/113629b1cf8769f65c09db0c3577a597e652860b/driver/include/esp_camera.h#L120-L121)" error. Looks like it can not detect it on I2C bus. Since it is a stock module, make sure you have set correct camera module in menuconfig:
ESP32 HomeKit Camera -> Camera Pins -> Select Camera Pinout -> ESP_EYE DevKit.

If that does not work, please post a photo of your quick reference card found inside packaging. It looks something like this:
https://makeradvisor.com/wp-content/uploads/2019/01/esp-eye-quick-guide.jpg
[image]

maximkulkin, 2019-09-16:
Oh, I haven't noticed the other error: ledc timer initialization failure. Looks like there is a problem with esp32-camera module which [still haven't caught up](https://github.com/espressif/esp32-camera/issues/66) with ESP-IDF changes. Try solution from there: put a `timer_conf.clk_cfg = LEDC_USE_APB_CLK;` line after [this line](https://github.com/espressif/esp32-camera/blob/master/driver/xclk.c#L23)

gozfree, 2019-09-17:
I have already select "ESP_EYE DevKit" but it does not work. And my quick reference card is the same with your two pictures.

maximkulkin, 2019-09-17:
@gozfree Read my later response then

gozfree, 2019-09-17:
Thanks! It works after adding clk_cfg.
But homekit seems crash after configuration:

> 

 >>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 4A:54:0E:6B:61:C4
>>> HomeKit: Found admin pairing with 07B9C109-6327-4B36-A381-01A6C3FB5521, disabling pair setup
>>> HomeKit: Configuring mDNS
>>> HomeKit: Got new client connection: 55
>>> HomeKit: [Client 55] Pair Verify Step 1/2
>>> HomeKit: [Client 55] Pair Verify Step 2/2
>>> HomeKit: [Client 55] Found pairing with 07B9C109-6327-4B36-A381-01A6C3FB5521
>>> HomeKit: [Client 55] Verification successful, secure session established
>>> HomeKit: [Client 55] Get Accessories
I (3384) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Resource
I (15744) esp32_camera: Resource payload: {"image-width":640,"image-height":480,"resource-type":"image"}
I (20624) wifi: ampdu: ignore deleting tx BA0
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Get Characteristics
>>> HomeKit: [Client 55] Update Characteristics
>>> HomeKit: [Client 55] Get Characteristics
I (23294) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 55] Update Characteristics
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40094b96  PS      : 0x00060033  A0      : 0x8008f15d  A1      : 0x3ffeb9e0  
A2      : 0x8f000000  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060020  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffeb9d0  
A10     : 0x3ffe0a08  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x0ccccccc  A15     : 0x00000057  SAR     : 0x00000001  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x8f000000  LBEG    : 0x4008dd58  LEND    : 0x4008dd63  LCOUNT  : 0x00000000  

ELF file SHA256: 381f731448a9083085739be6ee556fb1d60141d3d7eccdb0f73f6d8178080e22

Backtrace: 0x40094b93:0x3ffeb9e0 0x4008f15a:0x3ffeba10 0x40084e5d:0x3ffeba30 0x40084eaf:0x3ffeba50 0x40093065:0x3ffeba70 0x400d7bf9:0x3ffeba90 0x40093e79:0x3ffec2e0

Rebooting...
ets Jun  8 2016 00:22:57
`

>

maximkulkin, 2019-09-17:
Don't worry, [I've got you covered](https://github.com/maximkulkin/esp32-homekit-camera/commit/095a04facfff015c9e3155eb7beeda8a15922d85)!

Closing issue since original problem was solved

closed 2019-09-17 by maximkulkin