maximkulkin/esp32-homekit-camera #34: homekit qrcode

gozfree, 2019-09-21:
I modified the password and setupId in homekit_server_config，and re-generate the qrcode, but when I scan the qrcode by iphone homekit app, it prompt the device has already added, like "配件已添加"
I just bind the sample qrcode before and deleted the device. what should I do to bind the esp-eye with a new homekit qrcode?
Thanks!

AchimPieters, 2019-09-24:
Probably you need to reset the Homekit configuration, or reinstall the device. The Homekit apps tells you that it's already added.

maximkulkin, 2019-09-30:
When HomeKit server starts, it logs line like this:
```
mDNS announcement: Name=Camera-CD516A ␋md=LEDStrip␆pv=1.0␔id=E1:08:E4:1E:41:5D␄c#=1␄s#=1␄ff=0␄sf=1␄ci=5 
```
Please check that "sf" parameter is equal to "1". If it is "0", then pairing is disabled.

Also, right before that message it displays message like this:
```
HomeKit: Found admin pairing with 23E9459B-46A6-44E1-B35C-8E71AFAA324C, disabling pair setup
```

One thing to note, the way HomeKit works is: initial pairing is established by exchanging keys between accessory and controller (e.g. iOS device). That controller pairing becomes "admin" pairing. After that initial pairing is established, accessory does not accept any more pairings. Instead, "admin" controller should add other controller pairings via special API. The way it looks inside app is that you invite other people to control your home.

So, if you're trying to pair other iOS devices by scanning same QR code - that's not gonna work by design.

closed 2019-09-30 by maximkulkin

gozfree, 2019-10-07:
Thanks for your reply, I just using the same esp32 deivce and the same iOS phone for debugging homekit camera streaming, and now pairing is ok by commets some code.

maximkulkin, 2019-10-07:
I'm curious how your camera streaming debugging is going

gozfree, 2019-10-07:
camera streaming is still not work, blocked by malloc failed in camera_stream_task, 640*480*3 seems too large to malloc?