maximkulkin/esp32-homekit-camera #35: Compilation error

Kristian8606, 2019-09-21:
I don't understand why I get these errors?

> /Users/kristiandimitrov/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_init':
/Users/kristiandimitrov/esp/esp32-homekit-camera/components/homekit/src/storage.c:56:38: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
         strncpy(magic, magic1, sizeof(magic1));
                                      ^
In file included from /Users/kristiandimitrov/esp/esp32-homekit-camera/components/homekit/src/storage.c:6:
/Users/kristiandimitrov/esp/esp32-homekit-camera/components/homekit/src/storage.c: In function 'homekit_storage_can_add_pairing':
/Users/kristiandimitrov/esp/esp32-homekit-camera/components/homekit/src/port.h:20:85: warning: value computed is not used [-Wunused-value]
 #define spiflash_read(addr, buffer, size) (spi_flash_read((addr), (buffer), (size)) == ESP_OK)
  
camera/components/homekit/src/storage.c:233:39: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
     strncpy(data.magic, magic1, sizeof(magic1));

Staars, 2019-09-23:
If you do not want to turn off the error messages in the menuconfig, simply change the line to:
`strncpy(data.magic, magic1, sizeof(data.magic));`

You will run in to additional compilation errors after that, but all are fixable. But this is more or less cosmetic.

maximkulkin, 2019-09-23:
Those errors are already fixed in latest esp-homekit, but this repository wasn't yet updated. I recommend updating homekit component to latest master manually.

Kristian8606, 2019-09-23:
Аfter updating i get this one

> /Users/kristiandimitrov/esp/esp32-homekit-camera/components/libswscale/libswscale/output.c:2902:5: warning: enumeration value 'AV_PIX_FMT_YUVA444P12LE' not handled in switch [-Wswitch]
/Users/kristiandimitrov/esp/esp32-homekit-camera/components/libswscale/libswscale/output.c:2902:5: warning: enumeration value 'AV_PIX_FMT_NB' not handled in switch [-Wswitch]
cc1: some warnings being treated as errors
make[1]: *** [libswscale/output.o] Error 1
make: *** [component-libswscale-build] Error 2

Kristian8606, 2019-09-24:
the camera works!  I don't understand why the camera sends one image every 5-10 seconds.  is this the right way to work or am i confusing the settings

Kristian8606, 2019-09-24:
Get this error too

> E (725306) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (725306) task_wdt:  - IDLE1 (CPU 1)
E (725306) task_wdt: Tasks currently running:
E (725306) task_wdt: CPU 0: IDLE0
E (725306) task_wdt: CPU 1: HomeKit Server

AchimPieters, 2019-09-24:
> 
> 
> the camera works! I don't understand why the camera sends one image every 5-10 seconds. is this the right way to work or am i confusing the settings

This is fine, snapshot is requested by iOS, you can not change snapshot period.

AchimPieters, 2019-09-24:
> 
> 
> Get this error too
> 
> > E (725306) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> > E (725306) task_wdt:  - IDLE1 (CPU 1)
> > E (725306) task_wdt: Tasks currently running:
> > E (725306) task_wdt: CPU 0: IDLE0
> > E (725306) task_wdt: CPU 1: HomeKit Server

Streaming support was not implemented yet.

usman2006, 2019-09-25:
> 
> 
> the camera works! I don't understand why the camera sends one image every 5-10 seconds. is this the right way to work or am i confusing the settings

hi,
can you provide your source folder incl. "components" folder.
somehow i get always erros when i compile related to "components" folder and sometime it compiles through but in home app no pic. camera shows always no connected.

thx

Kristian8606, 2019-09-25:
I'll send you what you need, just give me an email

usman2006, 2019-09-26:
> 
> 
> I'll send you what you need, just give me an email

thx, 
i'v managed to get work.
as you pointed out...i just used old esp-idf and than it worked fine and now i see the pic.

but thx for offer.

Kristian8606, 2019-10-11:
After updating Mac OS Catalina, I get this error when trying to build:

> WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-80-g6c4433a
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.

Kristian8606, 2019-10-11:
and this 

> xtensa-esp32-elf-c++: internal compiler error: Killed: 9 (program cc1plus)

maximkulkin, 2019-10-11:
Whoa. I think it is time yo upgrade your esp-idf and toolchain

Kristian8606, 2019-10-11:
I updated the esp-idf to the latest version and now I'm getting this error

> cc1: some warnings being treated as errors
make[1]: *** [libswscale/output.o] Error 1
make: *** [component-libswscale-build] Error 2

Kristian8606, 2019-10-11:
I was able to build the project but now I have a hardware problem

> E (5438) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=3
E (5458) camera_xclk: ledc_timer_config failed, rc=ffffffff
I (5458) sccb: pin_sda 26 pin_scl 27

I (5468) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (5498) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (5558) camera: Detected camera not supported.
E (5558) camera: Camera probe failed with error 0x20004
E (5558) esp32_camera: Camera init failed with error 0x20004

maximkulkin, 2019-10-11:
You need to patch components/esp32-camera with [esp32-camera.patch](https://github.com/maximkulkin/esp32-homekit-camera/blob/master/esp32-camera.patch)

Kristian8606, 2019-10-11:
> Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40094932  PS      : 0x00060b33  A0      : 0x8008ecf0  A1      : 0x3ffc98d0  
0x40094932: uxPortCompareSet at /Users/kristiandimitrov/esp/esp-idf/components/freertos/include/freertos/portmacro.h:356
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /Users/kristiandimitrov/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/kristiandimitrov/esp/esp-idf/components/freertos/portmux_impl.h:99
 (inlined by) vTaskEnterCritical at /Users/kristiandimitrov/esp/esp-idf/components/freertos/tasks.c:4201

A2      : 0xee000000  A3      : 0x00000001  A4      : 0x00060b20  A5      : 0x0000cdcd  
A6      : 0xb33fffff  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffc9790  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffca6f8  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xee000000  LBEG    : 0x4008db40  LEND    : 0x4008db4b  LCOUNT  : 0x00000000

maximkulkin, 2019-10-11:
When did that happen?

maximkulkin, 2019-10-11:
I believe that you did not change HOMEKIT_SPI_FLASH_BASE_ADDR as it is described in README.md

Kristian8606, 2019-10-11:
after adding this line
    timer_conf.clk_cfg = LEDC_USE_APB_CLK;

Kristian8606, 2019-10-11:
I changed it

maximkulkin, 2019-10-11:
I mean, show the log before Guru meditation message

Kristian8606, 2019-10-11:
> I (1961) wifi: wifi firmware version: c01c829
I (1961) wifi: config NVS flash: enabled
I (1961) wifi: config nano formating: disabled
I (1961) wifi: Init dynamic tx buffer num: 32
I (1961) wifi: Init data frame dynamic rx buffer num: 32
I (1971) wifi: Init management frame dynamic rx buffer num: 32
I (1981) wifi: Init management short buffer num: 32
I (1981) wifi: Init static tx buffer num: 16
I (1981) wifi: Init static rx buffer size: 1600
I (1991) wifi: Init static rx buffer num: 10
I (1991) wifi: Init dynamic rx buffer num: 32
I (2091) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (2091) wifi: mode : sta (cc:50:e3:94:ec:f0)
I (2091) esp32_camera: STA start
I (3301) wifi: new:<10,2>, old:<1,0>, ap:<255,255>, sta:<10,2>, prof:1
I (3901) wifi: state: init -> auth (b0)
I (3901) wifi: state: auth -> assoc (0)
I (3911) wifi: state: assoc -> run (10)
I (4031) wifi: connected with Cveti, channel 10, 40D, bssid = 70:4f:57:8d:60:fc
I (4031) wifi: pm start, type: 1

I (5411) tcpip_adapter: sta ip: 192.168.0.118, mask: 255.255.255.0, gw: 192.168.0.1
I (5411) esp32_camera: WiFI ready
I (5411) esp32_camera: Free heap: 4364884
I (5421) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5421) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5431) sccb: pin_sda 26 pin_scl 27

I (5441) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (5491) sccb: SCCB_Probe start
I (6021) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6021) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6031) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6041) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6051) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6061) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6071) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6081) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6091) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6101) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6101) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6111) camera: Allocating 2 frame buffers (75 KB total)
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40094932  PS      : 0x00060b33  A0      : 0x8008ecf0  A1      : 0x3ffc98d0  
0x40094932: uxPortCompareSet at /Users/kristiandimitrov/esp/esp-idf/components/freertos/include/freertos/portmacro.h:356
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /Users/kristiandimitrov/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/kristiandimitrov/esp/esp-idf/components/freertos/portmux_impl.h:99
 (inlined by) vTaskEnterCritical at /Users/kristiandimitrov/esp/esp-idf/components/freertos/tasks.c:4201

A2      : 0xee000000  A3      : 0x00000001  A4      : 0x00060b20  A5      : 0x0000cdcd  
A6      : 0xb33fffff  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffc9790  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffca6f8  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xee000000  LBEG    : 0x4008db40  LEND    : 0x4008db4b  LCOUNT  : 0x00000000  
0x4008db40: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:133

0x4008db4b: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:143

ELF file SHA256: fdba58e868f823a3758d1f23d9d625d6315b9c5e3493dc975ec4246bc6f673bf

Backtrace: 0x4009492f:0x3ffc98d0 0x4008eced:0x3ffc9900 0x400847b9:0x3ffc9920 0x40084807:0x3ffc9940 0x40092e82:0x3ffc9960 0x40092ea4:0x3ffc9980 0x40152d43:0x3ffc99a0 0x40153558:0x3ffc99d0 0x40153851:0x3ffc9a00 0x400d7706:0x3ffc9a30 0x400d6a23:0x3ffc9ad0 0x400d6a68:0x3ffc9af0 0x40154f9d:0x3ffc9b10 0x401e928b:0x3ffc9b30 0x401e8e25:0x3ffc9b60 0x401e8ef5:0x3ffc9ba0 0x400930c1:0x3ffc9bc0
0x4009492f: uxPortCompareSet at /Users/kristiandimitrov/esp/esp-idf/components/freertos/include/freertos/portmacro.h:356
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /Users/kristiandimitrov/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/kristiandimitrov/esp/esp-idf/components/freertos/portmux_impl.h:99
 (inlined by) vTaskEnterCritical at /Users/kristiandimitrov/esp/esp-idf/components/freertos/tasks.c:4201

0x4008eced: multi_heap_internal_lock at /Users/kristiandimitrov/esp/esp-idf/components/heap/multi_heap.c:375
 (inlined by) multi_heap_malloc_impl at /Users/kristiandimitrov/esp/esp-idf/components/heap/multi_heap.c:416

[... 6 lines cut ...]
0x40092ea4: calloc at /Users/kristiandimitrov/esp/esp-idf/components/newlib/heap.c:37

0x40152d43: camera_fb_init at /Users/kristiandimitrov/esp/esp32-homekit-camera/components/esp32-camera/driver/camera.c:241

[... cut ...]
0x40154f9d: esp_event_post_to_user at /Users/kristiandimitrov/esp/esp-idf/components/esp_event/event_loop_legacy.c:34

0x401e928b: handler_execute at /Users/kristiandimitrov/esp/esp-idf/components/esp_event/esp_event.c:145

0x401e8e25: esp_event_loop_run at /Users/kristiandimitrov/esp/esp-idf/components/esp_event/esp_event.c:545 (discriminator 3)

0x401e8ef5: esp_event_loop_run_task at /Users/kristiandimitrov/esp/esp-idf/components/esp_event/esp_event.c:115

0x400930c1: vPortTaskWrapper at /Users/kristiandimitrov/esp/esp-idf/components/freertos/port.c:143

>

maximkulkin, 2019-10-11:
What is the output of `grep CONFIG_SPIRAM sdkconfig` in your setup?

Kristian8606, 2019-10-11:
sorry don't understand your question, maybe you ask about this setting?

> (X) Make RAM allocatable using malloc() as well            

>

maximkulkin, 2019-10-11:
Open sdkconfig file, find all lines that contain CONFIG_SPIRAM and post it here

Kristian8606, 2019-10-11:
> 
CONFIG_SPIRAM_BOOT_INIT=y
# CONFIG_SPIRAM_IGNORE_NOTFOUND is not set
# CONFIG_SPIRAM_USE_MEMMAP is not set
# CONFIG_SPIRAM_USE_CAPS_ALLOC is not set
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_TYPE_AUTO=y
# CONFIG_SPIRAM_TYPE_ESPPSRAM32 is not set
# CONFIG_SPIRAM_TYPE_ESPPSRAM64 is not set
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_SPIRAM_BANKSWITCH_ENABLE=y
CONFIG_SPIRAM_BANKSWITCH_RESERVE=8
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
# CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP is not set
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
# CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY is not set
# CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY is not set
CONFIG_D0WD_PSRAM_CLK_IO=17
CONFIG_D0WD_PSRAM_CS_IO=16
CONFIG_D2WD_PSRAM_CLK_IO=9
CONFIG_D2WD_PSRAM_CS_IO=10
CONFIG_PICO_PSRAM_CS_IO=10
CONFIG_SPIRAM_SPIWP_SD3_PIN=7

>

maximkulkin, 2019-10-11:
Which module you're loading that to?

Kristian8606, 2019-10-11:
https://www.aliexpress.com/item/33030892181.html?spm=a2g0o.productlist.0.0.105a4d3cPpUUzt&s=p&algo_pvid=789a4a98-a514-4a1f-95bd-400ca14abefa&algo_expid=789a4a98-a514-4a1f-95bd-400ca14abefa-0&btsid=d52eef0f-8bd5-4b8d-a0f6-324f55c55574&ws_ab_test=searchweb0_0,searchweb201602_10,searchweb201603_52

maximkulkin, 2019-10-11:
Hmm.. that's strange.. can you show log from the very start up until wifi init

Kristian8606, 2019-10-11:
`MONITOR
--- WARNING: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.SLAB_USBtoUART instead...
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6916
load:0x40078000,len:14852
ho 0 tail 12 room 4
load:0x40080400,len:4280
entry 0x400806e0
I (72) boot: Chip Revision: 1
I (73) boot_comm: mismatch chip revision, expect 1, found 0
I (40) boot: ESP-IDF v4.1-dev-369-g4dac7c7df 2nd stage bootloader
I (40) boot: compile time 01:47:48
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00200000
I (92) boot: End of partition table
I (96) boot_comm: mismatch chip revision, expect 1, found 0
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xa8938 (690488) map
I (358) esp_image: segment 1: paddr=0x000b8960 vaddr=0x3ffb0000 size=0x03d14 ( 15636) load
I (365) esp_image: segment 2: paddr=0x000bc67c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/kristiandimitrov/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (365) esp_image: segment 3: paddr=0x000bca84 vaddr=0x40080400 size=0x0358c ( 13708) load
I (380) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0x119314 (1151764) map
0x400d0018: _stext at ??:?

I (795) esp_image: segment 5: paddr=0x001d9334 vaddr=0x4008398c size=0x12e74 ( 77428) load
0x4008398c: psram_enable at /Users/kristiandimitrov/esp/esp-idf/components/esp32/spiram_psram.c:720

I (841) boot: Loaded app from partition at offset 0x10000
I (842) boot: Disabling RNG early entropy source...
I (842) psram: This chip is ESP32-D0WD
I (848) spiram: Found 64MBit SPI RAM device
I (851) spiram: SPI RAM mode: flash 40m sram 40m
I (856) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (864) cpu_start: Pro cpu up.
I (867) cpu_start: Application information:
I (872) cpu_start: Project name:     esp32-homekit-camera
I (878) cpu_start: App version:      a36e2d0-dirty
I (884) cpu_start: Compile time:     Oct 12 2019 01:47:55
I (890) cpu_start: ELF file SHA256:  2e9df6295db84841...
I (896) cpu_start: ESP-IDF:          v4.1-dev-369-g4dac7c7df
I (902) cpu_start: Starting app cpu, entry point is 0x40081420
0x40081420: call_start_cpu1 at /Users/kristiandimitrov/esp/esp-idf/components/esp32/cpu_start.c:276

I (0) cpu_start: App cpu up.
I (1794) spiram: SPI SRAM memory test OK
I (1794) heap_init: Initializing. RAM available for dynamic allocation:
I (1794) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1800) heap_init: At 3FFBBB10 len 000244F0 (145 KiB): DRAM
I (1807) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1813) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1820) heap_init: At 40096800 len 00009800 (38 KiB): IRAM
I (1826) cpu_start: Pro cpu start user code
I (1831) spiram: Adding pool of 4096K of external SPI memory to heap allocator
[... 97 lines cut ...]

0x40154f9d: esp_event_post_to_user at /Users/kristiandimitrov/esp/esp-idf/components/esp_event/event_loop_legacy.c:34

0x401e928b: handler_execute at /Users/kristiandimitrov/esp/esp-idf/components/esp_event/esp_event.c:145

0x401e8e25: esp_event_loop_run at /Users/kristiandimitrov/esp/esp-idf/components/esp_event/esp_event.c:545 (discriminator 3)

0x401e8ef5: esp_event_loop_run_task at /Users/kristiandimitrov/esp/esp-idf/components/esp_event/esp_event.c:115

0x400930c1: vPortTaskWrapper at /Users/kristiandimitrov/esp/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6916
load:0x40078000,len:14852
ho 0 tail 12 room 4
load:0x40080400,len:4280
entry 0x400806e0
`

Kristian8606, 2019-10-11:
i think the problem comes from mac os catalina because had to update esp-idf to the latest version and the problem came up

maximkulkin, 2019-10-12:
I’m running latest everything on latest Mac and do not have those problems. Although I’m using ESP-EYE module

Kristian8606, 2019-10-12:
what version of xcode command line tools are you with

Kristian8606, 2019-10-12:
after 3 minutes restart writes this 

> E (6508) sccb: SCCB_Write [ff]=01 failed
E (6508) sccb: SCCB_Write [12]=80 failed
I (6528) sccb: SCCB_Probe start
E (7788) camera: Camera probe failed with error 0x20001
E (7788) esp32_camera: Camera init failed with error 0x20001

Kristian8606, 2019-10-12:
the camera works with this arduino example
https://github.com/RuiSantosdotme/arduino-esp32-CameraWebServer

maximkulkin, 2019-10-14:
Ok, I've done tests on following camera modules I have:
* ESP-EYE
* M5Stack Camera 
* AI-Thinker ESP32-CAM

All of them worked: detected camera and was able to stream live video. It must be something with toolchain/settings. I will experiment with different toolchains and will get back.

Kristian8606, 2019-10-15:
when i compile with idf-3.3 version with the patch I get an error.  without patch I compile successfully but the esp restarts again

maximkulkin, 2019-10-15:
Did you do full clean before rebuilding (`make clean ; rm sdkconfig`)?

Kristian8606, 2019-10-15:
yes  and i do it all over again

Kristian8606, 2019-10-15:
I reinstalled my macbook and started over 
esp-idf v3.3 with patch

[image]

Kristian8606, 2019-10-16:
My question is stupid but how do  apply the patch? i tried windows but i got an error again

`C:/msys32/home/PC/esp/esp32-homekit-camera/components/esp32-camera/driver/xclk.c: In function 'camera_enable_out_clock':
C:/msys32/home/PC/esp/esp32-homekit-camera/components/esp32-camera/driver/xclk.c:23:15: error: 'ledc_timer_config_t {aka struct <anonymous>}' has no member named 'clk_cfg'
     timer_conf.clk_cfg = LEDC_USE_APB_CLK;
               ^
C:/msys32/home/PC/esp/esp32-homekit-camera/components/esp32-camera/driver/xclk.c:23:26: error: 'LEDC_USE_APB_CLK' undeclared (first use in this function)
     timer_conf.clk_cfg = LEDC_USE_APB_CLK;
                          ^
C:/msys32/home/PC/esp/esp32-homekit-camera/components/esp32-camera/driver/xclk.c:23:26: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [/home/PC/esp/esp-idf/make/component_wrapper.mk:290: driver/xclk.o] Error 1
make: *** [C:/msys32/home/PC/esp/esp-idf/make/project.mk:552: component-esp32-camera-build] Error 2
PC@DESKTOP-H0SQDFS MINGW32 ~/esp/esp32-homekit-camera
`

Kristian8606, 2019-10-19:
the restart issue seems to be caused by a misconfigured PIN on the LED
i changed it and now the camera works though with poor quality
it seems that after a few seconds of video streaming I get a problem with wdt

Kristian8606, 2019-10-19:
`E (204977) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (204977) task_wdt:  - IDLE0 (CPU 0)
E (204977) task_wdt: Tasks currently running:
E (204977) task_wdt: CPU 0: Camera Stream
E (204977) task_wdt: CPU 1: IDLE1
E (209977) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
`

gozfree, 2019-10-20:
> Ok, I've done tests on following camera modules I have:
> 
> * ESP-EYE
> * M5Stack Camera
> * AI-Thinker ESP32-CAM
> 
> All of them worked: detected camera and was able to stream live video.
> 
> It must be something with toolchain/settings. I will experiment with different toolchains and will get back.

I am testing on ESP-EYE, but it seems always crash and reboot repeatly when I use iPhone homekit to scan the QRcode to setup device. Could you help me to find the problem. I have commit the log to   your patch. Many thanks!

maximkulkin, 2019-10-20:
> I am testing on ESP-EYE, but it seems always crash and reboot repeatly when I use iPhone homekit to scan the QRcode to setup device. Could you help me to find the problem. I have commit the log to your patch. Many thanks!

I'm just letting you know that I ignore reports without any information. Uninstall. This is not for you.

usman2006, 2019-10-20:
> the restart issue seems to be caused by a misconfigured PIN on the LED
> i changed it and now the camera works though with poor quality
> it seems that after a few seconds of video streaming I get a problem with wdt

have same issue,
to what did you the nr. of the led pin on the led ?
(have the same esp32 with cam)

thx

Kowy54, 2019-10-20:
Makefile:5: /make/project.mk: No such file or directory
make: *** No rule to make target `/make/project.mk'.  Stop.
michael@Michael-MacBook-Pro esp32-homekit-camera-master % 
i have this problem if i try make menuconfig

maximkulkin, 2019-11-01:
You need to properly configure your SDK (esp-idf). Read installation instructions in eps-idf documentation.

gozfree, 2019-11-03:
I updated the latest code and using esp-idf v4.0-beta2 to compile, but seems error, could you help me, many thanks:

make[1]: *** No rule to make target 'http-parser/http_parser.o', needed by 'libhttp-parser.a'.  Stop.
/home/hacker/github/esp/esp-idf/make/project.mk:584: recipe for target 'component-http-parser-build' failed
make: *** [component-http-parser-build] Error 2

maximkulkin, 2019-11-05:
`make clean`

closed 2019-11-15 by maximkulkin