maximkulkin/esp32-homekit-camera #38: Runtime Error:esp_flash_write in esp_flash_api

GermanSheepDog, 2019-10-12:
Hello,

I download the new version of esp32-homekit-cam and compiled it successfully. The program started at esp32 and the connection to homekit started. At  Pair Setup Step 3/3 I got this error:
```
...
I (6142) camera: Allocating 2 frame buffers (75 KB total)
I (6152) camera: Allocating 37 KB frame buffer in OnBoard RAM
I (6162) camera: Allocating 37 KB frame buffer in OnBoard RAM
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: C0:B1:A8:09:BF:ED
>>> HomeKit: Configuring mDNS
>>> HomeKit: Got new client connection: 55
>>> HomeKit: [Client 55] Pair Setup Step 1/3
>>> HomeKit: [Client 55] Pair Setup Step 2/3
>>> HomeKit: [Client 55] Pair Setup Step 3/3
abort() was called at PC 0x400833d9 on core 0
0x400833d9: esp_flash_write at /Users/username/esp/esp-idf/components/spi_flash/esp_flash_api.c:609

ELF file SHA256: 259834b78efd7d0bfd0a74940c1c5bf4f4c8c4149ca0a202f5863d1e3bc23397

Backtrace: 0x4008bb55:0x3ffe6f50 0x4008bec5:0x3ffe6f70 0x400833d9:0x3ffe6f90 0x400e6bed:0x3ffe6fe0 0x4017f386:0x3ffe7000 0x4017e41f:0x3ffe7080 0x4017ef19:0x3ffe71d0 0x40183091:0x3ffe71f0 0x4017c757:0x3ffe7250 0x4017dc03:0x3ffe7280 0x4017ed32:0x3ffe72c0 0x4008e06d:0x3ffe72e0
0x4008bb55: invoke_abort at /Users/username/esp/esp-idf/components/esp32/panic.c:719

0x4008bec5: abort at /Users/username/esp/esp-idf/components/esp32/panic.c:719

0x400833d9: esp_flash_write at /Users/username/esp/esp-idf/components/spi_flash/esp_flash_api.c:609

0x400e6bed: spi_flash_write at /Users/username/esp/esp-idf/components/spi_flash/esp_flash_api.c:657

0x4017f386: homekit_storage_add_pairing at /Users/username/esp/esp32-homekit-camera/components/homekit/src/storage.c:247

0x4017e41f: homekit_server_on_pair_setup at /Users/username/esp/esp32-homekit-camera/components/homekit/src/server.c:1353

0x4017ef19: homekit_server_on_message_complete at /Users/username/esp/esp32-homekit-camera/components/homekit/src/server.c:2925

0x40183091: http_parser_execute at /Users/username/esp/esp32-homekit-camera/components/http-parser/http-parser/http_parser.c:2335

0x4017c757: homekit_client_process at /Users/username/esp/esp32-homekit-camera/components/homekit/src/server.c:3149

0x4017dc03: homekit_run_server at /Users/username/esp/esp32-homekit-camera/components/homekit/src/server.c:3257

0x4017ed32: homekit_server_task at /Users/username/esp/esp32-homekit-camera/components/homekit/src/server.c:3425

0x4008e06d: vPortTaskWrapper at /Users/username/esp/esp-idf/components/freertos/port.c:403

Rebooting...
I (12) boot: ESP-IDF v4.0-dev-1191-g138c941fa 2nd stage bootloader
I (12) boot: compile time 09:38:34
I (12) boot: Enabling RNG early entropy source...
I (17) boot: SPI Speed      : 40MHz
I (21) boot: SPI Mode       : DIO
I (25) boot: SPI Flash Size : 4MB
I (29) boot: Partition Table:
I (33) boot: ## Label            Usage          Type ST Offset   Length
I (40) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (48) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (55) boot:  2 factory          factory app      00 00 00010000 00200000
I (63) boot: End of partition table
I (67) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xa79d8 (686552) map
I (316) esp_image: segment 1: paddr=0x000b7a00 vaddr=0x3ffb0000 size=0x03e94 ( 16020) load
I (323) esp_image: segment 2: paddr=0x000bb89c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/username/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (323) esp_image: segment 3: paddr=0x000bbca4 vaddr=0x40080400 size=0x0436c ( 17260) load
I (339) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0x1017fc (1054716) map
0x400d0018: _stext at ??:?

I (710) esp_image: segment 5: paddr=0x001c181c vaddr=0x4008476c size=0x0c7ac ( 51116) load
0x4008476c: timer_initialized at /Users/username/esp/esp-idf/components/esp_common/src/ets_timer_legacy.c:47

I (742) boot: Loaded app from partition at offset 0x10000
I (742) boot: Disabling RNG early entropy source...
I (743) cpu_start: Pro cpu up.
... 
```
Can anyone help me? Which version of esp-idf did you use? Did you use Mac Catalina?

Bye,
Uwe

Kristian8606, 2019-10-12:
I use Catalina with this version
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/macos-setup.html

maximkulkin, 2019-10-12:
It looks like it has paired, but failed to write pairing info to flash. Check your flash size and address where homekit supposed to write it’s data.

GermanSheepDog, 2019-10-13:
Hello @maximkulkin,
hello @Kristian8606,
I think, there are amongst other things some problems with the versions of esp-idf.
Last time I used:
```
I (38) boot: ESP-IDF v4.1-dev-369-g4dac7c7df 2nd stage bootloader
```
and got this runtime errors:
```
I (5941) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (5951) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=3
E (5961) camera_xclk: ledc_timer_config failed, rc=ffffffff
I (5961) sccb: pin_sda 13 pin_scl 12

I (5971) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (6001) sccb: SCCB_Write [ff]=01 failed

E (6001) sccb: SCCB_Write [12]=80 failed

I (6021) sccb: SCCB_Probe start
E (7281) camera: Camera probe failed with error 0x20001
E (7281) esp32_camera: Camera init failed with error 0x20001
```
My previous version is:
```
I (884) cpu_start: ESP-IDF:          v4.0-dev-1191-g138c941fa
```
With this version I got this problem:
```
I (7089) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (7099) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (7109) camera: Allocating 2 frame buffers (75 KB total)
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40094eda  PS      : 0x00060533  A0      : 0x8008f52d  A1      : 0x3ffc9a90  
0x40094eda: uxPortCompareSet at /Users/username/esp/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /Users/username/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /Users/username/esp/esp-idf/components/freertos/portmux_impl.h:99
 (inlined by) vTaskEnterCritical at /Users/username/esp/esp-idf/components/freertos/tasks.c:4201
```
I suspect a problem with my hardware. I used a "TTGO T-Kamera ESP32 WROVER & PSRAM Kamera Modul ESP32-WROVER-B OV2640 Kamera Modul 0,96 OLED" and configured it as "TTGO CAM v1.7" with "Support for external, SPI connected RAM". It is correct?

Without "Support for external, SPI connected RAM" I got an runtime error, if I started video.

Which version of esp-idf do you use?
Which hardware do you use?

GermanSheepDog, 2019-10-15:
@maximkulkin: I like to isolate the problem and I will start one time more from scratch:
make clean; rm sdkconfig;
It will help me if I know the esp-idf version and the camera type which worked sure correctly. There are more unknown effects like MacOS Version and so on.

maximkulkin, 2019-10-15:
What is the problem with MacOS version?? You're using a separate toolchain, not bound neither to OS, nor to Xcode.

I tested code on ESP-IDF versions 3.3 (branch "v3.3" in Git), 4.0 (branch "release/v4.0") and on "master" (commit "4dac7c7df"). Tested it on ESP-EYE, M5Stack Camera and AI-Thinker ESP32-CAM modules. With v4.x you need to apply esp32-camera.patch to components/esp32-camera, with v3.x you do not need that. I did not test all permutations (mostly I was testing esp-idf master branch on different modules and then tested different esp-idf versions on ESP-EYE module). It worked in all configurations.

GermanSheepDog, 2019-10-18:
I did not pay enough attention to this message. Thank you!

I double-checked the two SPI RAM parameters in menuconfig:
```
ESP32-specific
 - Support for external, SPI-connected RAM = check 
 - SPI RAM config
    - Initialize SPI RAM when booting the ESP32 = check -> ok
    - SPI RAM access method = Make RAM allocatable using malloc() as well -> ok
```
At start of ESP32 it seemed the everything was ok:
```
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
**I (43) boot: SPI Flash Size : 4MB**
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00200000
I (81) boot: End of partition table
...
I (843) spiram: Found 64MBit SPI RAM device
I (846) spiram: SPI RAM mode: flash 40m sram 40m
I (852) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
...
I (889) cpu_start: App cpu up.
**I (1788) spiram: SPI SRAM memory test OK**
I (1788) heap_init: Initializing. RAM available for dynamic allocation:
I (1789) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1795) heap_init: At 3FFBBC80 len 00024380 (144 KiB): DRAM
I (1801) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1807) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1814) heap_init: At 400987F8 len 00007808 (30 KiB): IRAM
I (1820) cpu_start: Pro cpu start user code
I (1825) spiram: Adding pool of 4096K of external SPI memory to heap allocator
```
Is there any other Parameter in "SPI RAM config" to change? For example some parameter with memory size?

maximkulkin, 2019-10-18:
So where did all memory go?

GermanSheepDog, 2019-10-19:
Yes, it looks like I do have 4MB of RAM for HEAP.

I build in some logging information like this:
```
ESP_LOGI(TAG, "Free heap %s 1: %d", __FUNCTION__, xPortGetFreeHeapSize());
```
Output of 1. test:
```
I (7525) camera: Free heap camera_probe 2: 4397092
I (7525) camera_xclk: Free heap camera_enable_out_clock 1: 4397092
I (7535) camera_xclk: Free heap camera_enable_out_clock 2: 4397092
I (7545) camera_xclk: Free heap camera_enable_out_clock 3: 4397092
I (7545) camera_xclk: Free heap camera_enable_out_clock 4: 202828
I (7555) camera_xclk: Free heap camera_enable_out_clock 5: 202828
I (7565) camera: Free heap camera_probe 3: 202828
I (7565) sccb: pin_sda 22 pin_scl 23
```
Output of 2.  + 3. test:
```
I (7555) camera_xclk: Free heap camera_enable_out_clock 5: 4397092
I (7565) camera: Free heap camera_probe 3: 4397092
I (7565) sccb: pin_sda 22 pin_scl 23

I (7575) camera: Free heap camera_probe 4: 202416
```
Output of 4. test:
```
I (7527) camera: Free heap camera_probe 2: 4397092
I (7527) camera_xclk: Free heap camera_enable_out_clock 1: 4397092
I (7537) camera_xclk: Free heap camera_enable_out_clock 2: 4397092
I (7547) camera_xclk: Free heap camera_enable_out_clock 3: 202828
I (7547) camera_xclk: Free heap camera_enable_out_clock 4: 202828
```
Why are the process different?
Who stoled me 4MB of RAM?

GermanSheepDog, 2019-10-19:
I found the problem - it is the LED!
I use default parameters for LED ant it was mapped with GPIO16 (look at sdkconfig: CONFIG_CAMERA_PIN_LED=16)
I switched at my M5CameraF LED to Pin 2 (Data of Mic!) an it worked fine. I can see a stream,  a little bit blurred. Today I did not need the mic.

maximkulkin, 2019-10-19:
My M5Camera LED is GPIO 14, consult your documentation.

And congratulations!

closed 2019-10-19 by maximkulkin

GermanSheepDog, 2019-10-19:
Yes, in second step I found the LED on GPIO14.