maximkulkin/esp32-homekit-camera #39: Error while compiling

Electric-90, 2019-10-13:
I am using mingw32 on Windows as my development environment for esp-idf.
I was able to successfully compile the LED example for esp32 from esp-homekit-demo but I am getting the following error while trying to compile esp32-homekit-camera.

[image]

kenthinson, 2019-10-13:
I believe I am having the same issue. 

`CC examples/esp32-homekit-camera/build/x264/x264-snapshot-20181221-2245-stable/common/base.o
cc1: error: -Werror=format-truncation=: no option -Wformat-truncation=
make[1]: *** [x264-snapshot-20181221-2245-stable/common/base.o] Error 1`

GermanSheepDog, 2019-10-13:
I got the same problems, please look here: https://github.com/maximkulkin/esp32-homekit-camera/issues/38

Electric-90, 2019-10-13:
@GermanSheepDog I tried commenting out `#CFLAGS += -Wno-error=format-truncation=` line and the x264 packages compiled successfully but then I got stuck at the following error:

  
[image]

HomeACcessoryKid, 2019-10-13:
I got the same error when using the ESP-IDF = v3.3

when using v4.1-dev (which is what Max  seems to be using) I got a whole pile of errors related to spi_xxx

I think that for settling these issues we need to agree on a common version of esp_idf.

My personal favour would be v3.3 since it is LTS but Max has the ultimate say in this.

I also would like to know his macOS, his Xcode and xtensa version preferences

Any choice is valid, as long as people know what to aim for…

And please include the sdkconfig in the repository (get rid of the wifi-password though)

my 2 cents,
HacK

maximkulkin, 2019-10-14:
@HomeACcessoryKid A few people (including you) reported those problems with spi_xxx but surprisingly I have never seen those. Could you be more specific on what those errors are, which version of ESP-IDF you use and how do you build the project.

It feels that a lot of issues originate in different project settings, but I would prefer not to commit the whole sdkconfig as it would cement out settings that were never important and distract from what really important settings are. The way I do that (and recommend to anybody) is when you update an ESP-IDF, or esp32-homekit-camera, or any other component - you should start fresh: `make clean ; rm sdkconfig` After that, set values as described in README and it should work. I've done it with different ESP-IDF versions and different modules and it always worked for me.

As of supporting v3.3 (as being their LTS release) - it feels that v3.3 is the last in 3.x series and not surprisingly they promised to keep it for some time, but do you know what LTS in this even mean?

closed 2019-10-14 by Electric-90