maximkulkin/esp32-homekit-camera #40: M5Camera's

GermanSheepDog, 2019-10-19:
Hello,
there are some challenges with M5Camera's. There are the models
- M5Camera (A Model and B Model)
- M5CameraF
- M5CameraX (A Model and B Model)

The different between A model an B model are the camera pins SCCB Data and vertical sync.
I like to expand the KConfig and the camera.h file:
```
config CAMERA_MODEL_ESP_EYE
    bool "ESP_EYE DevKit"
config CAMERA_MODEL_M5_BFX_PSRAM
    bool "M5CameraB/F/X With PSRAM"
config CAMERA_MODEL_M5_A_PSRAM
    bool "M5CameraA With PSRAM"
config CAMERA_MODEL_AI_THINKER
    bool "ESP32-CAM by AI-Thinker"
```
```
#elif CONFIG_CAMERA_MODEL_M5_BFX_PSRAM
#define CAMERA_PIN_PWDN     -1
#define CAMERA_PIN_RESET    15
#define CAMERA_PIN_SIOD     22
#define CAMERA_PIN_SIOC     23
#define CAMERA_PIN_XCLK     27

#define CAMERA_PIN_D0       32
#define CAMERA_PIN_D1       35
#define CAMERA_PIN_D2       34
#define CAMERA_PIN_D3        5
#define CAMERA_PIN_D4       39
#define CAMERA_PIN_D5       18
#define CAMERA_PIN_D6       36
#define CAMERA_PIN_D7       19
#define CAMERA_PIN_VSYNC    25
#define CAMERA_PIN_HREF     26
#define CAMERA_PIN_PCLK     21

#elif CONFIG_CAMERA_MODEL_M5_A_PSRAM
#define CAMERA_PIN_PWDN     -1
#define CAMERA_PIN_RESET    15
#define CAMERA_PIN_SIOD       25
#define CAMERA_PIN_SIOC     23
#define CAMERA_PIN_XCLK     27

#define CAMERA_PIN_D0       32
#define CAMERA_PIN_D1       35
#define CAMERA_PIN_D2       34
#define CAMERA_PIN_D3        5
#define CAMERA_PIN_D4       39
#define CAMERA_PIN_D5       18
#define CAMERA_PIN_D6       36
#define CAMERA_PIN_D7       19
#define CAMERA_PIN_VSYNC      22
#define CAMERA_PIN_HREF     26
#define CAMERA_PIN_PCLK     21
```
In addition it would be nice if the LED-pin could be set in config.h and not separately in Kconfig.

The little different at the M5Camera's and the separately LED configuration cost me hours.

chrwh, 2021-09-23:
Can you share your code with this implementet?

rcarmo, 2022-07-02:
I too would like to see code for this. I have been looking at getting one of those cameras.