maximkulkin/esp32-homekit-camera #41: Crash using defined CAMERA_WIDTH and CAMERA_HEIGHT

ziolelle, 2019-11-01:
Hi, using the predefines values in config.h I experimented crash in camera initialization:

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40094b8e  PS      : 0x00060533  A0      : 0x8008eda8  A1      : 0x3ffc9c30  
0x40094b8e: uxPortCompareSet at /root/esp/esp-idf/components/freertos/include/freertos/portmacro.h:356
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /root/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /root/esp/esp-idf/components/freertos/portmux_impl.h:99
 (inlined by) vTaskEnterCritical at /root/esp/esp-idf/components/freertos/tasks.c:4201

A2      : 0xef000000  A3      : 0x00000001  A4      : 0x00060520  A5      : 0x0000cdcd  
A6      : 0xb33fffff  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffc98c0  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffcaa5c  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xef000000  LBEG    : 0x4008d50c  LEND    : 0x4008d517  LCOUNT  : 0x00000000  
0x4008d50c: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:133

0x4008d517: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:143

ELF file SHA256: 1a9724246109fbfb34eb1a949e08b13d0640941cf68dd18450cf429716d1ef9b

Backtrace: 0x40094b8b:0x3ffc9c30 0x4008eda5:0x3ffc9c60 0x400847c5:0x3ffc9c80 0x40084813:0x3ffc9ca0 0x4009306e:0x3ffc9cc0 0x40093090:0x3ffc9ce0 0x401611ef:0x3ffc9d00 0x40161a04:0x3ffc9d30 0x40161cfd:0x3ffc9d60 0x400d7586:0x3ffc9d90 0x400d67b7:0x3ffc9e30 0x400d67fc:0x3ffc9e50 0x401635a5:0x3ffc9e70 0x401f93ef:0x3ffc9e90 0x401f8f89:0x3ffc9ec0 0x401f9059:0x3ffc9f00 0x40093e39:0x3ffc9f20
0x40094b8b: uxPortCompareSet at /root/esp/esp-idf/components/freertos/include/freertos/portmacro.h:356
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /root/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /root/esp/esp-idf/components/freertos/portmux_impl.h:99
 (inlined by) vTaskEnterCritical at /root/esp/esp-idf/components/freertos/tasks.c:4201

0x4008eda5: multi_heap_internal_lock at /root/esp/esp-idf/components/heap/multi_heap.c:375
 (inlined by) multi_heap_malloc_impl at /root/esp/esp-idf/components/heap/multi_heap.c:416

0x400847c5: heap_caps_malloc at /root/esp/esp-idf/components/heap/heap_caps.c:111

0x40084813: heap_caps_malloc_default at /root/esp/esp-idf/components/heap/heap_caps.c:146

0x4009306e: _calloc_r at /root/esp/esp-idf/components/newlib/heap.c:73

0x40093090: calloc at /root/esp/esp-idf/components/newlib/heap.c:37

0x401611ef: camera_fb_init at /root/workspace/esp32-homekit-camera/components/esp32-camera/driver/camera.c:241

0x40161a04: camera_init at /root/workspace/esp32-homekit-camera/components/esp32-camera/driver/camera.c:1159

0x40161cfd: esp_camera_init at /root/workspace/esp32-homekit-camera/components/esp32-camera/driver/camera.c:1268

0x400d7586: camera_accessory_init at /root/workspace/esp32-homekit-camera/main/accessory.c:742

0x400d67b7: on_wifi_ready at /root/workspace/esp32-homekit-camera/main/app_main.c:18

0x400d67fc: event_handler at /root/workspace/esp32-homekit-camera/main/app_main.c:31 (discriminator 9)

0x401635a5: esp_event_post_to_user at /root/esp/esp-idf/components/esp_event/event_loop_legacy.c:34

0x401f93ef: handler_execute at /root/esp/esp-idf/components/esp_event/esp_event.c:145

0x401f8f89: esp_event_loop_run at /root/esp/esp-idf/components/esp_event/esp_event.c:545 (discriminator 3)

0x401f9059: esp_event_loop_run_task at /root/esp/esp-idf/components/esp_event/esp_event.c:115

0x40093e39: vPortTaskWrapper at /root/esp/esp-idf/components/freertos/port.c:143

```

I changed **config.h** with these values:

```
define CAMERA_FRAME_RATE 30
//#define CAMERA_FRAME_SIZE FRAMESIZE_VGA
#define CAMERA_FRAME_SIZE FRAMESIZE_QVGA

//#define CAMERA_WIDTH 640
//#define CAMERA_HEIGHT 480
#define CAMERA_WIDTH 320
#define CAMERA_HEIGHT 240

// #define VIDEO_WIDTH 160
// #define VIDEO_HEIGHT 120
#define VIDEO_WIDTH 80
#define VIDEO_HEIGHT 60

```

and changed also file **accessory.c** at line 755 and 756 disabling the comment on:
```
    tlv_add_integer_value(video_attributes, 1, 2, CAMERA_WIDTH);  // Image width
    tlv_add_integer_value(video_attributes, 2, 2, CAMERA_HEIGHT);  // Image height
//    tlv_add_integer_value(video_attributes, 1, 2, 640);  // Image width
//    tlv_add_integer_value(video_attributes, 2, 2, 480);  // Image height
    tlv_add_integer_value(video_attributes, 3, 1, CAMERA_FRAME_RATE);  // Frame rate

```

now the camera is working.
Is it ok or I missed something ?

ziolelle, 2019-11-01:
Sorry forgot to tell you that looking at the code, the error was generated trying to allocate memory for the framebuffers (original values were trying to allocate 76K).

maximkulkin, 2019-11-01:
Not sure why 640x480 does not work for you, it worked on 4 different models I tested on. What kind of module did you use? Did you follow configuration procedure described in README?

ziolelle, 2019-11-04:
Yes, I followed the README step by step.
The model is one found on Aliexpress: 
[ESP32-CAM WiFi Wireless module ESP32 Serial to WiFi ESP32 CAM SPI Flash Bluetooth Development Board with OV2640 Camera Module](https://www.aliexpress.com/item/32967142736.html?spm=a2g0s.9042311.0.0.27424c4dLwL3Gd)

Initially the model worked as a charm but a lot of time ago

maximkulkin, 2019-11-05:
Oh, I just noticed this:  to change resolution before it wasn't enough to just change it in that header file, you had to adjust other parameters in the code. I updated config.h, so that everything is in one place. Although, the crash because of that shouldn't have happened that early. Anyways, update the code and give it a try. And if that does not work, revert all your changes, you're not yet qualified to make them.

ziolelle, 2019-11-05:
Thanks I will test it

ziolelle, 2019-11-06:
Really thanks, it work with no further mods. 
Now however as I already wrote in the followup for issue #44 I am getting the same "coloured snow" in streaming mode.

Consider this issue closed

really thanks

closed 2019-11-06 by ziolelle