maximkulkin/esp32-homekit-camera #42: Streaming is it working now?

ziolelle, 2019-11-01:
Sorry, looking at the commits, I saw "live Streaming". Is it working now?
Trying it now I still have on my iPhone "Camera is not responding"

Am I missing something ?

thank you

maximkulkin, 2019-11-01:
It does work but requires some careful tuning.

closed 2019-11-01 by maximkulkin

alangoveia, 2019-11-15:
Hi @maximkulkin.
Can you, please, tell me where I can find this careful tuning?
I am using the latest esp-idf version.
thanks in advance

maximkulkin, 2019-12-02:
There is no established and foolproof way to do that yet, the settings in README work for me. If that does not work, then you need to investigate what's going on and I do not know how I can help with that. If you do not know how, I guess your only option is to wait. I personally waiting for a fix to ESP-IDF to be merged that should fix networking issues that sometimes occur. Might do some more code combing after that.