maximkulkin/esp32-homekit-camera #43: ESP-IDF 4.x without Patch

GermanSheepDog, 2019-11-04:
Hello Maxim,

today I started from scratch and download esp32-homekit-camera. Every time I reboot I have to enter many parameters and with ESP-IDF 4.x I have to run the patch.

Here is a solution without Patch. The file "*/driver/xclk.c" has to be changed:
```
//Add
#include <string.h>

//Add after this block:
timer_conf.duty_resolution = 2;
timer_conf.freq_hz = config->xclk_freq_hz;
timer_conf.speed_mode = LEDC_HIGH_SPEED_MODE;
timer_conf.timer_num = config->ledc_timer;
//New
if ( strstr( esp_get_idf_version(), "v4.") ) {
   timer_conf.clk_cfg = LEDC_USE_APB_CLK;
}
```

maximkulkin, 2019-11-05:
Not sure I understand "every time I reboot" part.

As of patch: it is provided separately, because esp32-camera is a submodule pointing to original esp32-camera repository maintained by Espressif. I do not want to fork it to fix this thing and I believe they will fix it eventually.

GermanSheepDog, 2019-11-06:
Sorry, "reboot" was not correct. I meant every time I "restarted" (development from scratch).

I think it makes not a big different for Espressif if you fix the mistake. It is possible to mark that there is a challenge.