maximkulkin/esp32-homekit-camera #44: Colored Snow

GermanSheepDog, 2019-11-04:
Hello Maxim,
I downloaded the last version and started from scratch.
I use a "M5Stack Camera with PSRAM" and the picture mode worked fine. If I started  stream, I got only colored snow:

[image]

Some days ago I tested the version with date 18.10.19. This stream worked a short time like you described it. The old version was ok. Did the M5Stack Camera with PSRAM" work at your new enviroment?

maximkulkin, 2019-11-05:
It looks like video image size misconfiguration: I can see your image on top (looks like it's lighter on left side and darker on right side), and then comes the garbage from memory. Before I start banging my head against the wall: did you change anything? Please show output of `git diff`

GermanSheepDog, 2019-11-06:
I didn't like it if you bang your head against a wall. I can help you, but today I didn't know what isn wrong.
Four weeks ago I expand your short installation guide:
```
$git clone --recursive https://github.com/maximkulkin/esp32-homekit-camera.git
Before compiling, you need to alter several settings in menuconfig:

$cd esp32-homekit-camera
$make menuconfig
•	Serial flasher
             Change/Check: Set serial Port  
                 For example: /dev/cu.SLAB_USBtoUART.
             Optional change: Default baud rate =  2MBaud
             Change Flash size = 4MB
•	ESP32 HomeKit Camera
              Set: WiFi SSID and WiFi Password
              Camera Pins
                     Change/Check: Select Camera Pinout = your variant of module
•	Partition Table
               Change Partition Table = Custom partition table CSV
               Check Custom partition CSV file = partitions.csv
•	Component config
o	ESP32-specific
             Change Support for external, SPI-connected RAM 
             SPI RAM config
                   Change: Initialize SPI RAM when booting the ESP32 
                   Check: SPI RAM access method = Make RAM allocatable using malloc() as well
o      Camera configuration
             Change OV2640 Support (only)
o	HomeKit
              Change: SPI flash address for storing HomeKit data = 0x3A0000
•	Do not forget to SAVE!
•	If using ESP-IDF 4.x:
              Add in components/driver/xclk.c after this line
              timer_conf.timer_num = config->ledc_timer;
              NEW: timer_conf.clk_cfg = LEDC_USE_APB_CLK;
```
I double-checked the guide and I used same M5Cam with PSRAM for old and current installation. Every time old installation worked fine and every time new installed showed "colored snow".

ziolelle, 2019-11-06:
the issue now also for me. no further modification has been made by me. only followed the same guide as GermanSheepDog
Picture mode is Ok, streaming is getting the same coloured snow

maximkulkin, 2019-11-07:
I was able to reproduce it, that was my silly mistake. Also, I've found out that supporting different resolutions is tricky, so for now I have changed it back to sticking with just 640x480.

Still in my tests the main problem is running out of internal WiFi buffers that freezes accessory completely.

closed 2019-11-07 by maximkulkin

GermanSheepDog, 2019-11-07:
@maximkulkin: Thank you, it worked fine with my M5Cam.

I keep my fingers crossed that you will detect the challenge with the WiFi buffers run over.