maximkulkin/esp32-homekit-camera #46: Couldn't compile

okanduzyel, 2019-11-19:
Hi,

I cloned esp32-homekit-camera example inside esp32-homekit-camera folder which I created with that path esp-homekit-demo/examples/esp32-homekit-camera

I cloned with that command -> git clone --recursive https://github.com/maximkulkin/esp32-homekit-camera.git .

Then, my direction like that:
	components
	main
	.gitignore
	.gitmodules
	LICENSE
	Makefile
	README.md	
	esp32-camera.patch
	
I tried to make menuconfig command got that:
make: *** test: No such file or directory. Stop. 

I tried same thing inside of main folder and got same respond. 

make -C test didn't work for me. make -C erase_flash either. What do I miss? Now, my linux machine can compile and flash any esp8266. Should I install anything for esp32?

AchimPieters, 2019-11-20:
https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP32

okanduzyel, 2019-11-20:
Hi, 

I tried them but I couldn't again. I am stuck in first step of https://docs.espressif.com/projects/esp-idf/en/latest/get-started/

I had solved that for esp8266 with this video https://www.youtube.com/watch?v=QBj8OLig8Kg

There is a vagrant file which contains line by line codes what It need to install. 

is there any vagrant file like above for esp32?

AchimPieters, 2019-11-20:
@okanduzyel maybe this will help:

https://www.studiopieters.nl/esp32-toolchain-installation/

okanduzyel, 2019-11-20:
Unfortunately, I am currently working on Windows computer and vagrant ubuntu. 🙁

AchimPieters, 2019-11-20:
Maybe this is a solution: https://www.studiopieters.nl/esp-homekit-sdk-ubuntu-on-virtualbox/

okanduzyel, 2019-11-20:
I wish that There were a solution as .bin files (for specific boards i.e ai thinker esp-cam), flash esp32 with appropriate .bin file for my board and at the end, connect it wifi like sonoff...

@maximkulkin

AchimPieters, 2019-11-21:
Welk maybe this is not for you.

okanduzyel, 2019-11-21:
Maybe, Maybe not.

Because, I did! This helped so much for me https://www.youtube.com/watch?v=EjfhrAFWVPM&t=379s

People who like me, is not a programmer, will live these problems. 

Please start to read that instructions -> https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html

Something are different that video. So you will deal with a few problem. when google the problem individually, you can find its solution. I dealt with more than 15-20 problems and now, I can erase-flash esp32.

Thank you for your helping @AchimPieters

closed 2019-11-21 by maximkulkin