maximkulkin/esp32-homekit-camera #47: Compile error

okanduzyel, 2019-11-21:
Hi maxim, 

I installed esp idf tools correctly and then, I can erase or flash esp32. I tried hello_world example of espif, it works. 

When I compile your esp32-homekit-camera example, I got that error:

error: incompatible type for argument 1 of 'camera_accessory_set_ip_address'
             camera_accessory_set_ip_address(event->event_info.got_ip.ip_info.ip);
                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
In file included from /home/vagrant/esp-homekit-demo/examples/esp32-homekit-camera/main/app_main.c:14:
/home/vagrant/esp-homekit-demo/examples/esp32-homekit-camera/main/accessory.h:7:49: note: expected 'ip4_addr_t' {aka 'struct ip4_addr'} but argument is of type 'esp_ip4_addr_t' {aka 'struct esp_ip4_addr'}
 void camera_accessory_set_ip_address(ip4_addr_t ip);

Could you check it please?

okanduzyel, 2019-11-22:
I uninstalled and installed again xtensa tools, deleted and cloned esp32-homekit-camera example. It still gives same error. But, esp32 espif examples works correctly. Could you check it please Maxim?

maximkulkin, 2019-11-22:
Just checked: mine builds just fine on latest esp-idf

okanduzyel, 2019-11-23:
Does it depend on menu configuration? Or board?

It also gives a lot of warnings. (Not used function etc.)

okanduzyel, 2019-11-24:
I tried it 3rd different computer and got same result. I install the esp idf from here;

https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html 

with building instructions. Cloning your example from here with; 

git clone -recursive https://github.com/maximkulkin/esp32-homekit-camera.git .

make menuconfig 

and making what you say about configuration requirements:

```
Partition Table
Partition Table = Custom partition table CSV
Custom partition CSV file = partitions.csv
Component config
ESP32-specific
Support for external, SPI-connected RAM = check
SPI RAM config
Initialize SPI RAM when booting the ESP32 = check
SPI RAM access method = Make RAM allocatable using malloc() as well
Camera configuration
OV2640 Support = check
HomeKit
SPI flash address for storing HomeKit data = 0x3A0000
ESP32 HomeKit Camera
WiFi SSID and WiFi Password
Camera Pins
Select Camera Pinout = your variant of module (AI-thinker)
```

And its response always that:

```
CC build/x264/x264-snapshot-20181221-2245-stable/encoder/lookahead.o
CC build/x264/x264-snapshot-20181221-2245-stable/encoder/macroblock.o
CC build/x264/x264-snapshot-20181221-2245-stable/encoder/me.o
CC build/x264/x264-snapshot-20181221-2245-stable/encoder/ratecontrol.o
CC build/x264/x264-snapshot-20181221-2245-stable/encoder/set.o
CC build/x264/x264-snapshot-20181221-2245-stable/encoder/slicetype-cl.o
AR build/x264/libx264.a
CC build/main/accessory.o
CC build/main/app_main.o
In file included from /home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c:8:
/home/vagrant/esp/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
/home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c: In function 'event_handler':
/home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c:30:77: error: incompatible type for argument 1 of 'camera_accessory_set_ip_address'
             camera_accessory_set_ip_address(event->event_info.got_ip.ip_info.ip);
                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
In file included from /home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c:14:
/home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/accessory.h:7:49: note: expected 'ip4_addr_t' {aka 'struct ip4_addr'} but argument is of type 'esp_ip4_addr_t' {aka 'struct esp_ip4_addr'}
 void camera_accessory_set_ip_address(ip4_addr_t ip);
                                      ~~~~~~~~~~~^~
/home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c: In function 'wifi_init':
/home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c:45:5: warning: 'tcpip_adapter_init' is deprecated [-Wdeprecated-declarations]
     tcpip_adapter_init();
     ^~~~~~~~~~~~~~~~~~
In file included from /home/vagrant/esp/esp-idf/components/esp_netif/include/esp_netif.h:31,
                 from /home/vagrant/esp/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/vagrant/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/vagrant/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from /home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c:6:
/home/vagrant/esp/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:30:6: note: declared here
 void tcpip_adapter_init(void)  __attribute__ ((deprecated));
      ^~~~~~~~~~~~~~~~~~
/home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c:46:5: warning: 'esp_event_loop_init' is deprecated [-Wdeprecated-declarations]
     ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
     ^~~~~~~~~~~~~~~
In file included from /home/vagrant/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/vagrant/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from /home/vagrant/esp/esp-idf/examples/esp32-homekit-camera/main/app_main.c:6:
/home/vagrant/esp/esp-idf/components/esp_event/include/esp_event_legacy.h:227:11: note: declared here
 esp_err_t esp_event_loop_init(system_event_cb_t cb, void *ctx) __attribute__ ((deprecated));
           ^~~~~~~~~~~~~~~~~~~
make[1]: *** [app_main.o] Error 1
make: *** [component-main-build] Error 2
vagrant@vagrant-ubuntu-trusty-64:~/esp/esp-idf/examples/esp32-homekit-camera$
```

okanduzyel, 2019-11-28:
@maximkulkin Could you share your all configurations of make menuconfig please?

Rafal74, 2019-12-02:
Hello Maxim,
I admire your projects very much.
I have same problem like @okanduzyel.
What do you advise to check ?
Thank You

fxday, 2019-12-03:
Me too. 

Should I need to patch esp32-camera.patch?

fxday, 2019-12-03:
> Me, too. @maximkulkin 
> 
> Should I need to patch esp32-camera.patch?
> 
> > Hello Maxim,
> > I admire your projects very much.
> > I have same problem like @okanduzyel.
> > What do you advise to check ?
> > Thank You

okanduzyel, 2019-12-17:
esp-idf v3.2 it works!

closed 2019-12-17 by okanduzyel

ryan99alero, 2019-12-26:
I went in on my own repo and commented out the original line and just edited the line to be what its looking for.  This allowed it to compile and flash.  Will test when I get home if the module actually works. 

// void camera_accessory_set_ip_address(ip4_addr_t ip);
void camera_accessory_set_ip_address(esp_ip4_addr_t ip);