maximkulkin/esp32-homekit-camera #49: Compiling error on macOS

okanduzyel, 2019-12-03:
Hi,

I have got that error on three different ubuntu machine: https://github.com/maximkulkin/esp32-homekit-camera/issues/47

So, I tried to compile it on macOS. Now, I have got that error: 

```
/Users/okanduzyel/esp/esp32-homekit-camera/components/esp32-camera/driver/xclk.c: In function 'camera_enable_out_clock':
/Users/okanduzyel/esp/esp32-homekit-camera/components/esp32-camera/driver/xclk.c:36:12: error: 'ledc_channel_config_t {aka struct <anonymous>}' has no member named 'hpoint'
     ch_conf.hpoint = 0;
            ^
make[1]: *** [driver/xclk.o] Error 1
make: *** [component-esp32-camera-build] Error 2
```

maximkulkin, 2019-12-03:
I think what is happening is Espressif changing their APIs in latest esp-idf. I suggest you stick with esp-idf 4.0 version.

okanduzyel, 2019-12-03:
> I think what is happening is Espressif changing their APIs in latest esp-idf. I suggest you stick with esp-idf 4.0 version.

which version do you prefer to or you can compile? I install it according to its web site and that web site contains always updated version.

okanduzyel, 2019-12-03:
https://docs.espressif.com/projects/esp-idf/en/v3.2.3/get-started/index.html

worked for me. Thanks a lot!

closed 2019-12-03 by okanduzyel

leovj, 2019-12-05:
Thank Maxim!
Do you mean 3.0 version?

Leo