maximkulkin/esp32-homekit-camera #50: esp32_camera: STA disconnected problem

okanduzyel, 2019-12-04:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6536
load:0x40078000,len:11320
load:0x40080400,len:7252
entry 0x40080780
[0;32mI (29) boot: ESP-IDF v3.2.3 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 02:44:59[0m
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
[0;32mI (33) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (39) boot: SPI Speed      : 40MHz[0m
[0;32mI (43) boot: SPI Mode       : QIO[0m
[0;32mI (47) boot: SPI Flash Size : 4MB[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (54) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (77) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (84) boot: End of partition table[0m
[0;32mI (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xaaed4 (700116) map[0m
[0;32mI (306) esp_image: segment 1: paddr=0x000baefc vaddr=0x3ffb0000 size=0x03d0c ( 15628) load[0m
[0;32mI (311) esp_image: segment 2: paddr=0x000bec10 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (313) esp_image: segment 3: paddr=0x000bf018 vaddr=0x40080400 size=0x00ff8 (  4088) load[0m
[0;32mI (323) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0x11a36c (1155948) map[0m
[0;32mI (675) esp_image: segment 5: paddr=0x001da38c vaddr=0x400813f8 size=0x16d20 ( 93472) load[0m
[0;32mI (723) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (723) boot: Disabling RNG early entropy source...[0m
[0;32mI (724) psram: This chip is ESP32-D0WD[0m
[0;32mI (729) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (733) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (738) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (745) cpu_start: Pro cpu up.[0m
[0;32mI (749) cpu_start: Starting app cpu, entry point is 0x40081448[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1639) spiram: SPI SRAM memory test OK[0m
[0;32mI (1639) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1640) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1646) heap_init: At 3FFBBB88 len 00024478 (145 KiB): DRAM[0m
[0;32mI (1652) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1659) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1665) heap_init: At 40098118 len 00007EE8 (31 KiB): IRAM[0m
[0;32mI (1671) cpu_start: Pro cpu start user code[0m
[0;32mI (1676) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (19) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (20) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
I (90) wifi: wifi driver task: 3ffcaeb8, prio:23, stack:3584, core=0
[0;32mI (90) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (90) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (120) wifi: wifi firmware version: 7997e4b
I (120) wifi: config NVS flash: enabled
I (120) wifi: config nano formating: disabled
I (120) wifi: Init dynamic tx buffer num: 32
I (130) wifi: Init data frame dynamic rx buffer num: 32
I (130) wifi: Init management frame dynamic rx buffer num: 32
I (140) wifi: Init management short buffer num: 32
I (140) wifi: Init static tx buffer num: 16
I (150) wifi: Init static rx buffer size: 1600
I (150) wifi: Init static rx buffer num: 10
I (150) wifi: Init dynamic rx buffer num: 32

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6536
load:0x40078000,len:11320
load:0x40080400,len:7252
entry 0x40080780
[0;32mI (29) boot: ESP-IDF v3.2.3 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 02:44:59[0m
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (39) boot: SPI Speed      : 40MHz[0m
[0;32mI (43) boot: SPI Mode       : QIO[0m
[0;32mI (47) boot: SPI Flash Size : 4MB[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (55) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (77) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (84) boot: End of partition table[0m
[0;32mI (89) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xaaed4 (700116) map[0m
[0;32mI (306) esp_image: segment 1: paddr=0x000baefc vaddr=0x3ffb0000 size=0x03d0c ( 15628) load[0m
[0;32mI (312) esp_image: segment 2: paddr=0x000bec10 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (313) esp_image: segment 3: paddr=0x000bf018 vaddr=0x40080400 size=0x00ff8 (  4088) load[0m
[0;32mI (323) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0x11a36c (1155948) map[0m
[0;32mI (675) esp_image: segment 5: paddr=0x001da38c vaddr=0x400813f8 size=0x16d20 ( 93472) load[0m
[0;32mI (723) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (723) boot: Disabling RNG early entropy source...[0m
[0;32mI (724) psram: This chip is ESP32-D0WD[0m
[0;32mI (729) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (733) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (738) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (745) cpu_start: Pro cpu up.[0m
[0;32mI (749) cpu_start: Starting app cpu, entry point is 0x40081448[0m
[0;32mI (741) cpu_start: App cpu up.[0m
[0;32mI (1639) spiram: SPI SRAM memory test OK[0m
[0;32mI (1640) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1640) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1646) heap_init: At 3FFBBB88 len 00024478 (145 KiB): DRAM[0m
[0;32mI (1653) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1659) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1666) heap_init: At 40098118 len 00007EE8 (31 KiB): IRAM[0m
[0;32mI (1672) cpu_start: Pro cpu start user code[0m
[0;32mI (1677) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (20) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (21) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
I (91) wifi: wifi driver task: 3ffcaeb8, prio:23, stack:3584, core=0
[0;32mI (91) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (91) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (121) wifi: wifi firmware version: 7997e4b
I (121) wifi: config NVS flash: enabled
I (121) wifi: config nano formating: disabled
I (121) wifi: Init dynamic tx buffer num: 32
I (131) wifi: Init data frame dynamic rx buffer num: 32
I (131) wifi: Init management frame dynamic rx buffer num: 32
I (141) wifi: Init management short buffer num: 32
I (141) wifi: Init static tx buffer num: 16
I (151) wifi: Init static rx buffer size: 1600
I (151) wifi: Init static rx buffer num: 10
I (151) wifi: Init dynamic rx buffer num: 32
[0;32mI (221) phy: phy_version: 4009, e7844c0, Jul 15 2019, 13:20:06, 0, 0[0m
I (221) wifi: mode : sta (24:6f:28:96:13:c0)
[0;32mI (221) esp32_camera: STA start[0m
[0;32mI (2271) esp32_camera: STA disconnected[0m
[0;32mI (4321) esp32_camera: STA disconnected[0m
[0;32mI (6371) esp32_camera: STA disconnected[0m
[0;32mI (8421) esp32_camera: STA disconnected[0m
[0;32mI (10471) esp32_camera: STA disconnected[0m
[0;32mI (12521) esp32_camera: STA disconnected[0m
[0;32mI (14571) esp32_camera: STA disconnected[0m
[0;32mI (16621) esp32_camera: STA disconnected[0m

```

Now, I have this issue. Could you share your make menuconfig configurations (all configurations)

AchimPieters, 2019-12-04:
You have a power issue Brownout = power issue

okanduzyel, 2019-12-04:
> You have a power issue Brownout = power issue

I have reset by hand. The first reset is that. Second reset about voltage dropping but I feeding it with laboratory power supply which can feed 5v 10a. Besides, Arduino I camera example works with that setup. 

What does it mean by “ esp32_camera: STA disconnected�[0m”?

okanduzyel, 2019-12-04:
> You have a power issue Brownout = power issue

@maximkulkin 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6632
load:0x40078000,len:11320
load:0x40080400,len:7288
entry 0x40080784
[0;32mI (29) boot: ESP-IDF v3.2.3 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 12:33:01[0m
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
[0;32mI (33) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (38) boot: SPI Speed      : 80MHz[0m
[0;32mI (43) boot: SPI Mode       : QIO[0m
[0;32mI (47) boot: SPI Flash Size : 4MB[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (54) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (77) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (84) boot: End of partition table[0m
[0;32mI (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xaad08 (699656) map[0m
[0;32mI (282) esp_image: segment 1: paddr=0x000bad30 vaddr=0x3ffb0000 size=0x03d04 ( 15620) load[0m
[0;32mI (287) esp_image: segment 2: paddr=0x000bea3c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (289) esp_image: segment 3: paddr=0x000bee44 vaddr=0x40080400 size=0x011cc (  4556) load[0m
[0;32mI (299) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0x11de0c (1170956) map[0m
[0;32mI (617) esp_image: segment 5: paddr=0x001dde2c vaddr=0x400815cc size=0x12d84 ( 77188) load[0m
[0;32mI (654) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (655) boot: Disabling RNG early entropy source...[0m
[0;32mI (655) psram: This chip is ESP32-D0WD[0m
[0;32mI (660) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (664) spiram: SPI RAM mode: flash 80m sram 40m[0m
[0;32mI (669) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (677) cpu_start: Pro cpu up.[0m
[0;32mI (680) cpu_start: Starting app cpu, entry point is 0x40081428[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1570) spiram: SPI SRAM memory test OK[0m
[0;32mI (1570) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1571) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1577) heap_init: At 3FFBBB70 len 00024490 (145 KiB): DRAM[0m
[0;32mI (1583) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1590) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1596) heap_init: At 40094350 len 0000BCB0 (47 KiB): IRAM[0m
[0;32mI (1602) cpu_start: Pro cpu start user code[0m
[0;32mI (1607) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (61) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (62) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
I (142) wifi: wifi driver task: 3ffcb6ec, prio:23, stack:3584, core=0
[0;32mI (142) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (142) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (162) wifi: wifi firmware version: 7997e4b
I (162) wifi: config NVS flash: enabled
I (162) wifi: config nano formating: disabled
I (162) wifi: Init dynamic tx buffer num: 32
I (172) wifi: Init data frame dynamic rx buffer num: 32
I (172) wifi: Init management frame dynamic rx buffer num: 32
I (182) wifi: Init management short buffer num: 32
I (182) wifi: Init static tx buffer num: 16
I (192) wifi: Init static rx buffer size: 1600
I (192) wifi: Init static rx buffer num: 10
I (202) wifi: Init dynamic rx buffer num: 32
[0;32mI (262) phy: phy_version: 4009, e7844c0, Jul 15 2019, 13:20:06, 0, 0[0m
I (262) wifi: mode : sta (24:6f:28:96:13:c0)
[0;32mI (262) esp32_camera: STA start[0m
[0;32mI (2312) esp32_camera: STA disconnected[0m
[0;32mI (4362) esp32_camera: STA disconnected[0m
[0;32mI (6412) esp32_camera: STA disconnected[0m
[0;32mI (8452) esp32_camera: STA disconnected[0m
[0;32mI (10502) esp32_camera: STA disconnected[0m
[0;32mI (12552) esp32_camera: STA disconnected[0m
[0;32mI (14602) esp32_camera: STA disconnected[0m
[0;32mI (16652) esp32_camera: STA disconnected[0m
[0;32mI (18692) esp32_camera: STA disconnected[0m
[0;32mI (20742) esp32_camera: STA disconnected[0m
[0;32mI (22792) esp32_camera: STA disconnected[0m
[0;32mI (24842) esp32_camera: STA disconnected[0m
[0;32mI (26882) esp32_camera: STA disconnected[0m
[0;32mI (28932) esp32_camera: STA disconnected[0m
```

I have changed something onmenuconfig but It still sticks that : esp32_camera: STA disconnected[0m

AchimPieters, 2019-12-04:
Poor quality USB cable;
USB cable is too long;
Board with some defect (bad solder joints);
Bad computer USB port;
Or not enough power provided by the computer USB port.

Google is your best friend: [Google](https://www.google.com/)

okanduzyel, 2019-12-04:
> Poor quality USB cable;
> USB cable is too long;
> Board with some defect (bad solder joints);
> Bad computer USB port;
> Or not enough power provided by the computer USB port.
> 
> Google is your best friend: [Google](https://www.google.com/)

I had read it. There is no usb cable. Power supply directly connected to esp-cam pins and it can support enough voltage and current. 

What does it mean "esp32_camera: STA disconnected�[0m"?

okanduzyel, 2019-12-04:
My wifi name has Turkish character "Ü" can it make a problem? 

It looks like that on configuration page

https://i.imgyukle.com/2019/12/04/Rs96tP.png

But I have written its true shape in the Kconfig.projbuild by hand. Nothing has been changed.

okanduzyel, 2019-12-04:
I have tested to connect different wifi name and password. It is same. 

> Poor quality USB cable;
> USB cable is too long;
> Board with some defect (bad solder joints);
> Bad computer USB port;
> Or not enough power provided by the computer USB port.
> 
> Google is your best friend: [Google](https://www.google.com/)

Now, It doesn't supply powered by usb cable but, as I said before, I programmed and tried many many times to Arduino camera example, It works.

Now, I connected it to lab. power supply. I don't think it is power problem.

okanduzyel, 2019-12-04:
Same setup, 
It can connect wifi, open camera and stream.

https://i.imgyukle.com/2019/12/04/RBJ2L6.png

`that is power issue : [0;33mW (306) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration[0m`

but my output has no phy init failed: 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6332
load:0x40078000,len:11624
load:0x40080400,len:6648
entry 0x40080764
[0;32mI (29) boot: ESP-IDF v3.3 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 16:19:23[0m
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
[0;32mI (33) boot: SPI Speed      : 40MHz[0m
[0;32mI (37) boot: SPI Mode       : DIO[0m
[0;32mI (41) boot: SPI Flash Size : 4MB[0m
[0;32mI (45) boot: Partition Table:[0m
[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (71) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (79) boot: End of partition table[0m
[0;32mI (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xb0d4c (724300) map[0m
[0;32mI (346) esp_image: segment 1: paddr=0x000c0d74 vaddr=0x3ffb0000 size=0x0421c ( 16924) load[0m
[0;32mI (353) esp_image: segment 2: paddr=0x000c4f98 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (354) esp_image: segment 3: paddr=0x000c53a0 vaddr=0x40080400 size=0x0ac70 ( 44144) load[0m
[0;32mI (380) esp_image: segment 4: paddr=0x000d0018 vaddr=0x400d0018 size=0x120210 (1180176) map[0m
[0;32mI (794) esp_image: segment 5: paddr=0x001f0230 vaddr=0x4008b070 size=0x0cc68 ( 52328) load[0m
[0;32mI (830) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (831) boot: Disabling RNG early entropy source...[0m
[0;32mI (831) psram: This chip is ESP32-D0WD[0m
[0;32mI (837) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (840) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (845) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (853) cpu_start: Pro cpu up.[0m
[0;32mI (856) cpu_start: Application information:[0m
[0;32mI (861) cpu_start: Project name:     esp32-homekit-camera[0m
[0;32mI (867) cpu_start: App version:      8ed186e[0m
[0;32mI (872) cpu_start: Compile time:     Dec  4 2019 16:19:29[0m
[0;32mI (878) cpu_start: ELF file SHA256:  d1eb299cd63de798...[0m
[0;32mI (884) cpu_start: ESP-IDF:          v3.3[0m
[0;32mI (889) cpu_start: Starting app cpu, entry point is 0x40081520[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1780) spiram: SPI SRAM memory test OK[0m
[0;32mI (1781) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1781) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1787) heap_init: At 3FFBC0A8 len 00023F58 (143 KiB): DRAM[0m
[0;32mI (1793) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1800) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1806) heap_init: At 40097CD8 len 00008328 (32 KiB): IRAM[0m
[0;32mI (1812) cpu_start: Pro cpu start user code[0m
[0;32mI (1817) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (161) cpu_start: Chip Revision: 1[0m
[0;33mW (162) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.[0m
[0;32mI (165) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (175) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
I (235) wifi: wifi driver task: 3ffcbd10, prio:23, stack:3584, core=0
[0;32mI (235) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (235) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (275) wifi: wifi firmware version: aeed694
I (275) wifi: config NVS flash: enabled
I (275) wifi: config nano formating: disabled
I (275) wifi: Init dynamic tx buffer num: 32
I (275) wifi: Init data frame dynamic rx buffer num: 32
I (285) wifi: Init management frame dynamic rx buffer num: 32
I (285) wifi: Init management short buffer num: 32
I (295) wifi: Init static tx buffer num: 16
I (295) wifi: Init static rx buffer size: 1600
I (295) wifi: Init static rx buffer num: 10
I (305) wifi: Init dynamic rx buffer num: 32
[0;32mI (405) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0[0m
I (405) wifi: mode : sta (24:6f:28:96:13:c0)
[0;32mI (405) esp32_camera: STA start[0m
[0;32mI (2465) esp32_camera: STA disconnected[0m
[0;32mI (4515) esp32_camera: STA disconnected[0m
[0;32mI (6565) esp32_camera: STA disconnected[0m
[0;32mI (8625) esp32_camera: STA disconnected[0m
[0;32mI (10675) esp32_camera: STA disconnected[0m
[0;32mI (12725) esp32_camera: STA disconnected[0m
[0;32mI (14785) esp32_camera: STA disconnected[0m
[0;32mI (16835) esp32_camera: STA disconnected[0m
[0;32mI (18885) esp32_camera: STA disconnected[0m
[0;32mI (20945) esp32_camera: STA disconnected[0m

```

@maximkulkin  could you share your esp idf, xtensa and other tools versions please?

I can't understand it is hardware or software issue.

leovj, 2019-12-04:
I have the same problem...

leovj, 2019-12-04:
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6876
load:0x40078000,len:14876
ho 0 tail 12 room 4
load:0x40080400,len:4284
entry 0x400806e0
[0;32mI (72) boot: Chip Revision: 1[0m
[0;32mI (73) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (41) boot: ESP-IDF v4.1-dev-474-g2e6398aff 2nd stage bootloader[0m
[0;32mI (41) boot: compile time 11:16:38[0m
[0;32mI (41) boot: Enabling RNG early entropy source...[0m
[0;32mI (48) boot: SPI Speed      : 40MHz[0m
[0;32mI (52) boot: SPI Mode       : DIO[0m
[0;32mI (56) boot: SPI Flash Size : 4MB[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (86) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (93) boot: End of partition table[0m
[0;32mI (97) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xa9fa4 (696228) map[0m
[0;32mI (362) esp_image: segment 1: paddr=0x000b9fcc vaddr=0x3ffb0000 size=0x04290 ( 17040) load[0m
[0;32mI (369) esp_image: segment 2: paddr=0x000be264 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (370) esp_image: segment 3: paddr=0x000be66c vaddr=0x40080400 size=0x019a4 (  6564) load[0m
[0;32mI (381) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0x12c010 (1228816) map[0m
[0;32mI (826) esp_image: segment 5: paddr=0x001ec030 vaddr=0x40081da4 size=0x16438 ( 91192) load[0m
[0;32mI (880) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (880) boot: Disabling RNG early entropy source...[0m
[0;32mI (881) psram: This chip is ESP32-D0WD[0m
[0;32mI (886) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (890) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (895) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (902) cpu_start: Pro cpu up.[0m
[0;32mI (906) cpu_start: Application information:[0m
[0;32mI (911) cpu_start: Project name:     esp32-homekit-camera[0m
[0;32mI (917) cpu_start: App version:      8ed186e[0m
[0;32mI (922) cpu_start: Compile time:     Nov 23 2019 11:16:44[0m
[0;32mI (928) cpu_start: ELF file SHA256:  673def5ffa317b81...[0m
[0;32mI (934) cpu_start: ESP-IDF:          v4.1-dev-474-g2e6398aff[0m
[0;32mI (940) cpu_start: Starting app cpu, entry point is 0x400814c8[0m
[0;32mI (926) cpu_start: App cpu up.[0m
[0;32mI (1832) spiram: SPI SRAM memory test OK[0m
[0;32mI (1833) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1833) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1839) heap_init: At 3FFBC098 len 00023F68 (143 KiB): DRAM[0m
[0;32mI (1845) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1852) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1858) heap_init: At 400981DC len 00007E24 (31 KiB): IRAM[0m
[0;32mI (1864) cpu_start: Pro cpu start user code[0m
[0;32mI (1869) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1891) spi_flash: detected chip: generic[0m
[0;32mI (1892) spi_flash: flash io: dio[0m
[0;32mI (1892) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (1900) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
I (1980) wifi: wifi driver task: 3ffcbb34, prio:23, stack:3584, core=0
[0;32mI (1980) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (1980) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (2010) wifi: wifi firmware version: 932678a
I (2010) wifi: config NVS flash: enabled
I (2010) wifi: config nano formating: disabled
I (2010) wifi: Init dynamic tx buffer num: 32
I (2010) wifi: Init data frame dynamic rx buffer num: 32
I (2020) wifi: Init management frame dynamic rx buffer num: 32
I (2030) wifi: Init management short buffer num: 32
I (2030) wifi: Init static tx buffer num: 16
I (2030) wifi: Init static rx buffer size: 1600
I (2040) wifi: Init static rx buffer num: 10
I (2040) wifi: Init dynamic rx buffer num: 32

Brownout detector was triggered

https://i.imgyukle.com/2019/12/04/RBJ2L6.png

leovj, 2019-12-04:
I have tried whit external power supply 5v 1A.. Same problem..

https://i.imgyukle.com/2019/12/04/RBJ2L6.png

okanduzyel, 2019-12-16:
Is there anyone who can correctly run it on esp32?

okanduzyel, 2019-12-17:
It connect while there is no Turkish characters on your wifi SSID!

closed 2019-12-17 by okanduzyel

DenisMaldonadoC, 2022-03-13:
I was having the same error. In my case it was a stupid error. In the WiFi SSID section I wasn’t erasing the “myWifi” you have to deleted everything and only leave the name of your WiFi.

than-sad, 2023-02-26:
Hi, in my case was because my SSID in router is Home and I wrote HOME in menuconfig.