maximkulkin/esp32-homekit-camera #51: Crash camera

okanduzyel, 2019-12-17:
Camera shows pictures in each 10 seconds on favorite page (main page of homekit)

When I press the camera view to see stream, it gives that output and crashes:

```
[0;32mI (437617) esp32_camera: Resource payload: {"image-width":640,"image-height":480,"resource-type":"image"}[0m
>>> HomeKit: [Client 56] Resource
[0;32mI (439657) esp32_camera: Resource payload: {"image-width":640,"image-height":480,"resource-type":"image"}[0m
>>> HomeKit: [Client 57] Get Characteristics
>>> HomeKit: [Client 57] Update Characteristics
>>> HomeKit: [Client 57] Get Characteristics
[0;32mI (440787) esp32_camera: Creating setup endpoints response[0m
>>> HomeKit: [Client 57] Update Characteristics
[0;32mI (440997) esp32_camera: Processing session command 1[0m
[0;32mI (440997) esp32_camera: Adding streaming session[0m
[0;32mI (440997) esp32_camera: Starting streaming[0m
[0;32mI (440997) esp32_camera: Total free memory: 4203648[0m
x264 [info]: using cpu capabilities: none!
x264 [info]: profile Constrained Baseline, level 1.0
[0;32mI (441097) esp32_camera: Executing streaming loop[0m
[0;32mI (441107) esp32_camera: Total free memory: 3635344[0m
[0;32mI (441107) esp32_camera: Getting a frame[0m
[0;32mI (441107) esp32_camera: Total free memory: 3635344[0m
[0;32mI (441117) esp32_camera: JPEG output size: 80x60[0m
[0;32mI (441167) esp32_camera: JPEG finishing[0m
[0;32mI (441167) esp32_camera: JPEG destroying[0m
[0;32mI (441167) esp32_camera: Returning frame buffer[0m
[0;32mI (441167) esp32_camera: Done grabbing a frame[0m
[0;32mI (441167) esp32_camera: Encoding a frame[0m
[0;32mI (441177) esp32_camera: Total free memory: 3635252[0m
[0;32mI (441387) esp32_camera: Encoded frame, size = 2495[0m
[0;32mI (441387) esp32_camera: Sending RTCP sender report[0m
[0;32mI (441407) esp32_camera: Done sending frame data[0m
[0;32mI (441427) esp32_camera: Getting a frame[0m
[0;32mI (441427) esp32_camera: Total free memory: 3550928[0m
[0;32mI (441427) esp32_camera: JPEG output size: 80x60[0m
[0;32mI (441477) esp32_camera: JPEG finishing[0m
[0;32mI (441477) esp32_camera: JPEG destroying[0m
[0;32mI (441477) esp32_camera: Returning frame buffer[0m
[0;32mI (441477) esp32_camera: Done grabbing a frame[0m
[0;32mI (441477) esp32_camera: Encoding a frame[0m
[0;32mI (441487) esp32_camera: Total free memory: 3550928[0m
[0;32mI (441557) esp32_camera: Encoded frame, size = 25[0m
[0;32mI (441557) esp32_camera: Done sending frame data[0m
[0;32mI (441577) esp32_camera: Getting a frame[0m
[0;32mI (441577) esp32_camera: Total free memory: 3426096[0m
[0;32mI (441577) esp32_camera: JPEG output size: 80x60[0m
[0;32mI (441627) esp32_camera: JPEG finishing[0m
[0;32mI (441627) esp32_camera: JPEG destroying[0m
[0;32mI (441627) esp32_camera: Returning frame buffer[0m
[0;32mI (441627) esp32_camera: Done grabbing a frame[0m
[0;32mI (441637) esp32_camera: Encoding a frame[0m
[0;32mI (441637) esp32_camera: Total free memory: 3426096[0m
[0;32mI (441727) esp32_camera: Encoded frame, size = 29[0m
[0;32mI (441727) esp32_camera: Done sending frame data[0m
[0;32mI (441747) esp32_camera: Getting a frame[0m
[0;32mI (441747) esp32_camera: Total free memory: 3426096[0m
[0;32mI (441747) esp32_camera: JPEG output size: 80x60[0m
[0;32mI (441797) esp32_camera: JPEG finishing[0m
[0;32mI (441797) esp32_camera: JPEG destroying[0m
[0;32mI (441797) esp32_camera: Returning frame buffer[0m
[0;32mI (441797) esp32_camera: Done grabbing a frame[0m
[0;32mI (441807) esp32_camera: Encoding a frame[0m
[0;32mI (441807) esp32_camera: Total free memory: 3426096[0m
[0;32mI (441897) esp32_camera: Encoded frame, size = 51[0m
[0;32mI (441907) esp32_camera: Done sending frame data[0m
[0;32mI (441917) esp32_camera: Getting a frame[0m
[0;32mI (441917) esp32_camera: Total free memory: 3425716[0m
[0;32mI (441917) esp32_camera: JPEG output size: 80x60[0m
[0;32mI (442017) esp32_camera: JPEG finishing[0m
[... 197 lines cut ...]
[0;32mI (446487) esp32_camera: Done sending frame data[0m
[0;32mI (446497) esp32_camera: Getting a frame[0m
[0;32mI (446497) esp32_camera: Total free memory: 3424580[0m
[0;32mI (446497) esp32_camera: JPEG output size: 80x60[0m
[0;32mI (446547) esp32_camera: JPEG finishing[0m
[0;32mI (446547) esp32_camera: JPEG destroying[0m
[0;32mI (446547) esp32_camera: Returning frame buffer[0m
[0;32mI (446547) esp32_camera: Done grabbing a frame[0m
[0;32mI (446557) esp32_camera: Encoding a frame[0m
[0;32mI (446557) esp32_camera: Total free memory: 3424580[0m
[0;32mI (446647) esp32_camera: Encoded frame, size = 59[0m
[0;32mI (446657) esp32_camera: Done sending frame data[0m
[0;31mE (446657) esp32_camera: Failed to send encrypted packet (code 12)[0m
[0;31mE (446657) esp32_camera: Failed to send RTP packet (code -2)[0m
x264 [info]: frame I:3     Avg QP:17.67  size:  2472
x264 [info]: frame P:20    Avg QP:20.75  size:   294
x264 [info]: mb I  I16..4: 100.0%  0.0%  0.0%
x264 [info]: mb P  I16..4:  0.2%  0.0%  0.0%  P16..4: 73.2%  0.0%  0.0%  0.0%  0.0%    skip:26.5%
x264 [info]: coded y,uvDC,uvAC intra: 100.0% 100.0% 98.4% inter: 44.2% 56.9% 27.1%
x264 [info]: i16 v,h,dc,p: 10% 49% 41%  0%
x264 [info]: i8c dc,h,v,p: 51% 31%  7% 11%
x264 [info]: kb/s:115.66
[0;32mI (446707) esp32_camera: Done with stream[0m
ets Jun  8 2016 00:22:57
```

closed 2019-12-18 by maximkulkin

jp112sdl, 2019-12-22:
Hi!

Same issue here... any suggestions?

maximkulkin, 2019-12-23:
Wait for this to be fixed: https://github.com/espressif/esp-idf/issues/4309

doabigcheese, 2020-01-11:
@maximkulkin With latest IDF (ESP-IDF v4.1-dev-1865-gca979fb11) i get different log output when trying to stream:
I (2241509) esp32_camera: Processing session command 1 I (2241509) esp32_camera: Adding streaming session I (2241509) esp32_camera: Starting streaming I (2241519) esp32_camera: Total free memory: 2347188 x264 [info]: using cpu capabilities: none! x264 [error]: malloc of size 4957118 failed E (2243739) esp32_camera: Failed to open H.264 encoder E (2243739) FreeRTOS: FreeRTOS Task "Camera Stream" should not return, Aborting now! abort() was called at PC 0x40099b43 on core 0

Is this the same issue (with just other/better log output) or a different problem?
at least it is now recovering by itselve with a reboot :)

thnx