maximkulkin/esp32-homekit-camera #52: How to apply patch

okanduzyel, 2019-12-23:
Hi,

When I try to apply your patch file, I have got that output.

```
okanduzyel@MacBook-Pro esp32-homekit-camera % patch < esp32-camera.patch 
can't find file to patch at input line 5
Perhaps you should have used the -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/driver/xclk.c b/driver/xclk.c
|index 89ced66..21db2fc 100755
|--- a/driver/xclk.c
|+++ b/driver/xclk.c
--------------------------
File to patch: 
```

Could you show me How I can apply that patch?

maximkulkin, 2019-12-23:
`patch -p 1 < esp32-camera.patch`

closed 2019-12-23 by maximkulkin

MnemonicFish, 2023-01-16:
not working for me 

`c:\Users\Amein\esp\esp32-homekit-camera>patch -p 1 < esp32-camera.patch
'patch' is not recognized as an internal or external command,
operable program or batch file.`

MnemonicFish, 2023-01-16:
patch -p 1 < esp32-camera.patch  

---------------- does not work  

c:\Users\Amein\esp\esp32-homekit-camera>patch -p 1 < esp32-camera.patch
'patch' is not recognized as an internal or external command,
operable program or batch file.