maximkulkin/esp32-homekit-camera #54: Module doesn't boot and resets

haeckse, 2019-12-27:
With a AI-Thinker module, menuconfig options set according to the 'readme', the software won't boot. 
Looking at the partition lengths shown during boot, the values set in partitions.csv seem to be applied, but the errors are related to the partitioning.
I have built the firmware with esp-idf v4.1.
Do I need to change offsets/sizes in partitions.csv?

> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:7184
> ho 0 tail 12 room 4
> load:0x40078000,len:14816
> load:0x40080400,len:4456
> entry 0x400806e8
> I (120) boot: Chip Revision: 1
> I (138) boot_comm: chip revision: 1, min. bootloader chip revision: 0
> I (82) boot: ESP-IDF v4.1-dev-1572-g30372f5a4 2nd stage bootloader
> I (82) boot: compile time 17:11:21
> I (82) boot: Enabling RNG early entropy source...
> I (89) boot: SPI Speed      : 40MHz
> I (93) boot: SPI Mode       : DIO
> I (97) boot: SPI Flash Size : 4MB
> I (101) boot: Partition Table:
> I (104) boot: ## Label            Usage          Type ST Offset   Length
> I (112) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (119) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (127) boot:  2 factory          factory app      00 00 00010000 00200000
> I (135) boot: End of partition table
> I (139) boot_comm: chip revision: 1, min. application chip revision: 0
> I (146) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xac534 (705844) map
> I (407) esp_image: segment 1: paddr=0x000bc55c vaddr=0x3ffb0000 size=0x03abc ( 15036) load
> I (414) esp_image: segment 2: paddr=0x000c0020 vaddr=0x400d0020 size=0x136b9c (1272732) map
> 0x400d0020: _stext at ??:?
> 
> I (869) esp_image: segment 3: paddr=0x001f6bc4 vaddr=0x3ffb3abc size=0x0096c (  2412) load
> I (871) esp_image: segment 4: paddr=0x001f7538 vaddr=0x40080000 size=0x00400 (  1024) load
> 0x40080000: _WindowOverflow4 at /home/andrea/Projects/esp/esp-idf/components/freertos/xtensa_vectors.S:1778
> 
> I (876) esp_image: segment 5: paddr=0x001f7940 vaddr=0x40080400 size=0x1ca20 (117280) load
> E (934) esp_image: Image length 2114448 doesn't fit in partition length 2097152
> E (934) boot: Factory app partition is not bootable
> E (936) boot: No bootable app partitions in the partition table
> ets Jun  8 2016 00:22:57

okanduzyel, 2020-01-08:
- Try to use esp-idf v3.2
- Don't patch. 
- make appropriate menuconfig settings that mentioned in readme

It works. (without streaming)

haeckse, 2020-01-09:
@okanduzyel With esp-idf v3.2 it worked instantly without any issues. I get a new snapshot every 10s in the main view of the HK app, and even a live stream when I switch to the camera details view.

Thanks a lot for your help! :)

closed 2020-01-09 by haeckse

okanduzyel, 2020-01-09:
Doesn’t the camera crash when try to see its live stream?

maximkulkin, 2020-01-09:
I tested yesterday on latest ESP-IDF: it kind of works, did ~10 live streams, never crashed. There is one problem though: sometimes for some reason streaming task does not exit after stream is finished and keeps spinning and subsequent live streams do not work. Haven't looked into that yet.

okanduzyel, 2020-01-09:
This’s is nice news indeed! I can’t wait to see its live streaming working

haeckse, 2020-01-09:
This is exactly what happened during my tests today also. Sometimes the streaming-mode does not end when switching back to snapshot view. One can see the streaming task continues running in the monitor output then, while the camera connection ist lost.