maximkulkin/esp32-homekit-camera #55: Changing camera frame dimensions the live function not working

zonehun, 2020-01-15:
Hello Maxim,

I got a strange issue and I can't solve it. If I change these values in config.h

#define CAMERA_FRAME_RATE 30
#define CAMERA_FRAME_SIZE FRAMESIZE_VGA
#define CAMERA_WIDTH 640
#define CAMERA_HEIGHT 480

to these
#define CAMERA_FRAME_RATE 20
#define CAMERA_FRAME_SIZE FRAMESIZE_SVGA
#define CAMERA_WIDTH 800
#define CAMERA_HEIGHT 600

The still image is OK and the picture changes every 10sec. But if I click on the image to watch the live stream the Home app displays the "No response. This camera is not responding." message. 

There are no errors in the log in the same time. 

Do you have any suggestions how to solve this and change the dimensions to bigger ones?

(ps. the streaming working as expected with the 640x480 values.)

Thanks

maximkulkin, 2020-01-15:
Changing resolution is not supported

closed 2020-01-15 by maximkulkin

zonehun, 2020-01-16:
> Changing resolution is not supported

Is it not supported by the esp32-homekit-camera project or some of the dependecies not supported it? (for eg. eps32-home)

maximkulkin, 2020-01-16:
By this project. Live streaming involves heavy computation, and ESP32 is not fast enough to keep up at normal resolutions. Settings were tweaked for at least something bearable.

zonehun, 2020-01-16:
I see. I try to do some sort of things and if I have any results I will share it with you.

I like this little cam very much and when I used it through Homebrdige it was worked in bigger resolutions as well. So I hope there are some tweaking possibilities or some workaround... Anyway your project is great and I like the approach to be able to use the ESP's functions under Home app  without Homebridge.

cgmckeever, 2020-07-19:
@zonehun any luck with this? My unit seems to restart a bunch

zonehun, 2020-07-19:
> @zonehun any luck with this? My unit seems to restart a bunch

Unfortunately nothing.  :(