maximkulkin/esp32-homekit-camera #56: how to install/flash/compile

Pairan, 2020-01-16:
I've been lurking for long to get my hardware and since it has arrived, I would like to flash it with your software.

Is there an easy way to install it on my ESP32-CAM?

maximkulkin, 2020-01-17:
This is the only way:
1. Follow these build instructions: https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP32
2. Make configurations changes described in [README.md](https://github.com/maximkulkin/esp32-homekit-camera/blob/master/README.md)

Pairan, 2020-01-17:
Thank you so  much - I will try it and report back in

mrTuomoK, 2020-02-09:
@Pairan Did you get it working, I'm in the exact same situation myself trying to get it working...

cgmckeever, 2020-07-21:
@Pairan @mrTuomoK The [README](https://github.com/maximkulkin/esp32-homekit-camera) has been updated with some instructions. I recently got it up, def. some hiccups, but its more the learning curve than anything else

ystrem, 2021-01-09:
Hello guys, I have issue with the configuration step, I installed Prerequisites. When I run make menuconfig, I get 

x@mx:~/Desktop/esp32-homekit-camera
$ make menuconfig
Makefile:5: /make/project.mk: No such file or directory
make: *** No rule to make target '/make/project.mk'.  Stop.