maximkulkin/esp32-homekit-camera #57: error: 'rtc_gpio_desc' undeclared

alangoveia, 2020-01-18:
I am having a trouble with the error: 'etc_gpio_desc'

`App "esp32-homekit-camera" version: 2a26d79
CC build/esp32-camera/driver/twi.o
/home/ubuntucamera/esp/esp32-homekit-camera/components/esp32-camera/driver/twi.c: In function 'pinMode':
/home/ubuntucamera/esp/esp32-homekit-camera/components/esp32-camera/driver/twi.c:61:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
                        ^~~~~~~~~~~~~
                        rtc_io_desc
/home/ubuntucamera/esp/esp32-homekit-camera/components/esp32-camera/driver/twi.c:61:24: note: each undeclared identifier is reported only once for each function it appears in
/home/ubuntucamera/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'driver/twi.o' failed
make[1]: *** [driver/twi.o] Error 1
/home/ubuntucamera/esp/esp-idf/make/project.mk:604: recipe for target 'component-esp32-camera-build' failed
make: *** [component-esp32-camera-build] Error 2`

I executed the command make menuconfig and set the parameters.
when I try to flash i got the error above.
Anyone experienced that?
thanks

Millhouse273, 2020-01-18:
I#m having the same problem. Trying all night to get it running with no sucess :-(

maximkulkin, 2020-01-18:
There is a setting on latest ESP-IDF you need to enable for this project. I’m afk now, but you should be able to find it somewhere in component configs -> device configurations -> rtc something. Just enable it and it will compile.

Millhouse273, 2020-01-19:
Thanks alot for your answer. I found it and now my camera is online. Great project by the way!

alangoveia, 2020-01-19:
thanks a lot Maxim
It is in Component config > driver configurations > RTCI0 configuration

thanks again

closed 2020-01-19 by alangoveia

andreacantadori, 2021-01-12:
Thank you, extremely useful, I was stuck on this for 1 hour...

krambriw, 2021-06-17:
I have the same problem, fighting with this. Arduino w ESP support added. First under Linux, now on Windows, with this project:
https://www.hackster.io/anatoli-arkhipenko/multi-client-mjpeg-streaming-from-esp32-47768f

Same problem both on Linux as Windows
`exit status 1
'rtc_gpio_desc' undeclared (first use in this function)
`
`twi.c uses rtc_gpio_desc and it requires Component config > driver configurations > RTCI0 configuration > "Support array 'rtc_gpio_desc' for ESP32
`
I do not use make menuconfig so is there another way I can fix this, a file somewhere to edit?

Sarah-C, 2022-09-03:
@krambriw  Same here :(

than-sad, 2023-02-26:
Component config > driver configurations > RTCI0 configuration this option is missing in the actual ESP-IDF 5.0.1

PR #70 (2020-07-19): updated getting started instructions and expose additional settings via menuconfig
Three main themes:

* Improve getting started instructions. Resolves #57 and #58, makes note of #68  
* Feature: configurations of below settings via **menuconfig**
   * allow image rotation (vertical and horizontal) 
   * image enhancement (saturation and contrast)
* Enhancement: expose previously hardcoded HomeKit related settings into **menuconfig**
   * settings such as: setup code, manufacturer name, model name and number, serial number, and firmware version