maximkulkin/esp32-homekit-camera #58: Partitions flash size

pocketscience, 2020-02-03:
After updating to the specified configurations there was an error

`Partitions defined in '<blah>/esp32-homekit-camera/partitions.csv' occupy 2.1MB of flash (2162688 bytes) which does not fit in configured flash size 2MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.`

I'm using a DIYMore module which apparently has 4MB of flash, so changing the config to specify 4MB in the config fixed the issue.

Maybe mention that flash memory size might/will have to be adjusted as well.

jwktje, 2020-02-20:
I came across this. But I found that the output of the CLI gives enough direction. It tells you exactly what to change, and where to do it.

pocketscience, 2020-02-20:
Yup, but it would be easy to avoid this error by mentioning this upfront. Not the end of the world, granted, but an easy issue to avoid so why not fix it.