maximkulkin/esp32-homekit-camera #59: Secure video support?

nixmeer, 2020-02-07:
Hello :) 
Does this support the HomeKit Secure Video feature? It started with iOS 13.2 (I think)
Best :)

jwktje, 2020-02-11:
Was wondering about this too. No idea how feasible this is but it would be very awesome

maximkulkin, 2020-02-11:
No, it does not support Secure Video, mainly because it is a fairly complex piece of functionality (to try to reverse engineer) and it is not documented anywhere.

nixmeer, 2020-02-11:
Isn’t HomeKit documented on the Apple developer site? How do the companies do it that offer these devices?

maximkulkin, 2020-02-11:
You can easy open HomeKit spec published by Apple and fine that all out. What Apple opened to broad public is noncommercial HomeKit spec. Which means that there is a "commercial" version. Although, noncommercial version seem to be pretty accurate, but it seems that it is lagging behind in terms of adding new services/characteristics (and other facts).

Companies that develop commercial devices have gone through special Apple certification, signed NDA and got access to commercial SDKs and specs. I've seen [people doing reverse engineering of secure video HomeKit profile](https://github.com/Supereg/secure-video-specification), but they are not complete for implementation purposes.

nixmeer, 2020-02-16:
Thank you! Well I hope someone will publish it one day...

closed 2020-02-16 by nixmeer

korgoth, 2023-03-20:
Seems there is now a homebridge plugin ([https://github.com/SeydX/homebridge-camera-ui](https://github.com/SeydX/homebridge-camera-ui)) that has somehow implemented secure video. Does anyone have any idea of how they have done it and if that is somehow possible to implement in this project?