maximkulkin/esp32-homekit-camera #60: Having hard to get built & flashed, can I do this in docker? or easy way to do it in MacOS?

mrTuomoK, 2020-02-08:
I'm successful to build & flash the esp-homedit-demo examples with the docker-method so could this be done almost the same way? If not, does anyone have bullet-proof chain of commands to build & flash in macos environment please/thanks?

mrTuomoK, 2020-02-10:
Nevermind, got it working after some tweaking :)

closed 2020-02-10 by mrTuomoK

cgmckeever, 2020-07-19:
@mrTuomoK Would you mind sharing some of your steps to use this library? Im looking for a howto/tutorial but not coming up with much . .thank you

hire-vladimir, 2020-07-19:
@mrTuomoK take a look at the latest https://github.com/maximkulkin/esp32-homekit-camera/blob/master/README.md, i've updated it to hopefully clear up the process.

cgmckeever, 2020-07-19:
@hire-vladimir Thanks -- working on some other things with esp homekit, but will def. try this out soon .

cgmckeever, 2020-07-19:
@hire-vladimir could you expand on what 
```
Before compiling, you need to alter several settings in menuconfig:
```

This means? Such as maybe the /path/to/file to edit? or something?

cgmckeever, 2020-07-19:
Maybe this is the hint, its a config screen in the makefile? 
```
Configure device settings per menuconfig settings section, make menuconfig
```

?

hire-vladimir, 2020-07-19:
yes, it's the config screen of makefile. Step 4 gave the command of how to get to it `make menuconfig`, but i see how it can be confusing.

peros550, 2020-07-21:
I would like to make a question in regards to the prerequisites. This project uses the esp-idf. If I add esp-idf in the the path, would that conflict in any way to esp-homekit-demo which as far as I understand uses the esp-open-rtos?

cgmckeever, 2020-07-21:
@peros550 not an authority here, but these are used at compile time. If you manage the compile environments correctly, this should not be an issue, as once they are compiled and flashed thats no longer relevent

I have both installed on my laptop, and everything compiles and flashes as needed 

You could always remove something from the path, or have it only in a shell - but having it in the path usually doesnt do anything unless you clobber another entry

peros550, 2020-07-21:
@cgmckeever thank you for your support! I followed the instructions and was successful flashing my first esp32 camera :)

cgmckeever, 2020-07-21:
@peros550 I may hit you up to get `esp-homekit-demo` working ;) ..