maximkulkin/esp32-homekit-camera #62: Compiled binary for instant flashing

nixmeer, 2020-03-20:
Hello :) is it possible to add a working binary to this repo that everyone can upload to the esp32 using the esptool.py? 
- is it possible at all to upload a esp idf binary using the esptool.py?
- are all parameters configurable after flashing? Or do I have to add things like ssid and the WiFi password before compiling?
Best :)
Jan

pocketscience, 2020-03-26:
The wifi config is set during make menuconfig so i don't think a pre-made binary is an option unless you were to move from a static wifi setup to something like wifi_config. Are you having trouble building a binary on your own? What errors are you seeing?

nixmeer, 2020-03-26:
Well, I don't have the tool chain and idf installed yet and I try not do so. What other options are set there? Is it just wifi config to other settings as well?

pocketscience, 2020-03-26:
There's a few other settings required such as camera module type, portion table etc. You really need to install the tools to use it...

nixmeer, 2020-03-26:
A college of mine uses it at work, so we might do it there. But maybe a server, a little GUI and the thing opening up a wifi hotspot if it can't connect to a wifi network would be great. If corona stays around long enough, I might take a look at it

pocketscience, 2020-03-26:
Sounds like a great project! Go for it!

closed 2020-04-02 by nixmeer