maximkulkin/esp32-homekit-camera #63: Rotation

icefabio, 2020-04-09:
Is there any way to rotate the image? 

Thanks

HerrHerrmannMann, 2020-04-29:
you must change some hex-values.
path: /esp32-camera/sensors/private_include/ov2640_regs.h
look at line:153
---original---------
#define REG04_DEFAULT       0x28
#define REG04_HFLIP_IMG     0x80
#define REG04_VFLIP_IMG     0x40
#define REG04_VREF_EN       0x10
#define REG04_HREF_EN       0x08
#define REG04_SET(x)        (REG04_DEFAULT|x)
---original----------
change the hex-values between "#define REG04_DEFAULT" and "#define REG04_VFLIP_IMG"

that was vor me the easiest way :)

HerrHerrmannMann, 2020-04-29:
add two lines in /esp32-homekit-camera/compenents/esp32-camera/driver/camera.c

at line 1221 starts the ov2640 sensor.id.PID settings. 
add under the "s_state->sensor.set_bpc(&s_state->sensor, false);" line this:

s_state->sensor.set_hmirror(&s_state->sensor, true);
s_state->sensor.set_vflip(&s_state->sensor, true);

icefabio, 2020-05-01:
Hi Herr! Many thanks for your tip. :)

Assuming these 2 lines rotate the image 180 degrees, how could I rotate it 90 or 270 degrees? What's the rule?

HerrHerrmannMann, 2020-05-16:
hey,

I´ve looking for your issue. Rotation is not possible. Only hmirror & vflip.
You can rotate the image with a software solution. But, in this case, i´m out. I´m a bad coder...

icefabio, 2020-05-16:
Hi! It's clear :)

Tks

closed 2020-05-16 by icefabio