maximkulkin/esp32-homekit-camera #67: Question: RTSP streaming possible?

peros550, 2020-06-15:
I would like to give this firmware a try. 

Despite the homekit functionallity, does it expose RTSP stream? So that someone could view the streaming outside of the homekit environment?

cgmckeever, 2020-07-22:
Do you need the homekit parts at all? There are some simpler utilities that just stream to a browser on the same network

maximkulkin, 2020-07-22:
@peros550 as far as I understand, in addition to RTSP stream itself you need a different way to establish connection, negotiate streaming config, exchange encryption keys, etc. My goal was not to build a comprehensive library to plug into any occasion, but to build a finished IP camera firmware. Now that it is (kind of) working and have all it's limitations identified, I do not feel like it is of much use (since it can deliver only very low res stream) and it is worth investing time in it.
I was curious to get a real camera and reverse engineer security features (like video recording).

peros550, 2020-07-22:
@cgmckeever I would like to use home app as an easy, simplified and standard method to have access to all my home's accessories. Having all cameras in home app eliminates the need for having to open different apps depending on the camera's maker. 

Having said that, I find RTSP a convenient supplementary method to stream video to other devices. For example, I have setup a Synology NAS which allows me to record video on a 24x7. Having an RTPS makes that possible.

@maximkulkin thank you for your feedback. I wasn't aware of the complexity of this idea. Once again, thank you for your efforts.

closed 2020-07-22 by peros550