maximkulkin/esp32-homekit-camera #68: Home App not finding the camera

sam632, 2020-06-18:
hi, was wondering if anyone could help me. I have an ESP32 Cam and tried flashing this to it. The first time it seemed to work and the home app found the accessory but wouldn't connect to it. 

[image]

Then it disappeared from the adding accessory page so I reflashed it and now it won't find it at all. Serial monitor gives the following,

```c
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6176
ho 0 tail 12 room 4
load:0x40078000,len:10180
load:0x40080400,len:6640
entry 0x40080760
I (31) boot: ESP-IDF v3.2 2nd stage bootloader
I (31) boot: compile time 19:14:35
I (31) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00200000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x6aae0 (436960) map
I (247) esp_image: segment 1: paddr=0x0007ab08 vaddr=0x3ffb0000 size=0x03d1c ( 15644) load
I (253) esp_image: segment 2: paddr=0x0007e82c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/samadamson/esp32/esp-idf/components/freertos/xtensa_vectors.S:1779

I (254) esp_image: segment 3: paddr=0x0007ec34 vaddr=0x40080400 size=0x013dc (  5084) load
I (264) esp_image: segment 4: paddr=0x00080018 vaddr=0x400d0018 size=0x11bd40 (1162560) map
0x400d0018: _stext at ??:?

I (678) esp_image: segment 5: paddr=0x0019bd60 vaddr=0x400817dc size=0x18618 ( 99864) load
0x400817dc: esp_dport_access_stall_other_cpu_start at /Users/samadamson/esp32/esp-idf/components/esp32/dport_access.c:107

I (735) boot: Loaded app from partition at offset 0x10000
I (735) boot: Disabling RNG early entropy source...
I (737) spiram: Found 64MBit SPI RAM device
I (740) spiram: SPI RAM mode: flash 40m sram 40m
I (746) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (753) cpu_start: Pro cpu up.
I (757) cpu_start: Starting app cpu, entry point is 0x40081478
0x40081478: call_start_cpu1 at /Users/samadamson/esp32/esp-idf/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
I (1638) spiram: SPI SRAM memory test OK
I (1638) heap_init: Initializing. RAM available for dynamic allocation:
I (1638) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1645) heap_init: At 3FFBBBB8 len 00024448 (145 KiB): DRAM
I (1651) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1657) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1664) heap_init: At 40099DF4 len 0000620C (24 KiB): IRAM
I (1670) cpu_start: Pro cpu start user code
I (1675) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (19) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (20) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (110) wifi: wifi driver task: 3ffcb78c, prio:23, stack:3584, core=0
I (110) wifi: wifi firmware version: 9415913
I (110) wifi: config NVS flash: enabled
I (110) wifi: config nano formating: disabled
I (120) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (130) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (160) wifi: Init dynamic tx buffer num: 32
I (160) wifi: Init data frame dynamic rx buffer num: 32
I (160) wifi: Init management frame dynamic rx buffer num: 32
I (160) wifi: Init static tx buffer num: 16
I (170) wifi: Init static rx buffer size: 1600
I (170) wifi: Init static rx buffer num: 10
I (180) wifi: Init dynamic rx buffer num: 32
I (240) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 0
I (250) wifi: mode : sta (c8:2b:96:a0:a0:2c)
I (250) esp32_camera: STA start
I (370) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1500) wifi: state: init -> auth (b0)
I (1500) wifi: state: auth -> assoc (0)
I (1510) wifi: state: assoc -> run (10)
I (1520) wifi: connected with Wi-Fi, channel 1
I (1520) wifi: pm start, type: 1

I (6110) event: sta ip: 192.168.1.122, mask: 255.255.255.0, gw: 192.168.1.254
I (6110) esp32_camera: WiFI ready
I (6110) esp32_camera: Free heap: 4364552
I (6110) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6120) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6130) sccb: pin_sda 26 pin_scl 27

I (6140) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (6170) sccb: SCCB_Write Failed addr:0x3c, reg:0xff, data:0x01, ret:-1
E (6170) sccb: SCCB_Write Failed addr:0x3c, reg:0x12, data:0x80, ret:-1
I (6260) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6260) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6270) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6280) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6290) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6300) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6310) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6320) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6320) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6330) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6340) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6350) camera: Allocating 2 frame buffers (75 KB total)
I (6360) camera: Allocating 37 KB frame buffer in OnBoard RAM
I (6380) camera: Allocating 37 KB frame buffer in OnBoard RAM
>>> HomeKit: Starting server
I (6570) esp32_camera: Initializing streaming
>>> HomeKit: Using existing accessory ID: F2:B2:D1:E2:E8:F9
>>> HomeKit: Found admin pairing with BFED9AD6-A1D3-45B8-AB6B-E58F71B4A19A, disabling pair setup
I (6570) esp32_camera: Streaming initialized
>>> HomeKit: Configuring mDNS
```

evansj, 2020-07-14:
I think I have the same issue - I see from the logs that it connects to the correct WiFi network, DHCP succeeds, it detects an OV2640 camera, but ultimately fails at the `HomeKit: Configuring mDNS` stage. With a bit more debugging enabled I can see that it tries to start an http server:

```
>>> HomeKit: Starting server
I (3647) esp32_camera: Initializing streaming
>>> HomeKit: Formatting HomeKit storage at 0x3a0000
I (3687) esp32_camera: Streaming initialized
>>> HomeKit: Generated new accessory ID: xx:xx:xx:xx:xx:xx
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
>>> homekit_run_server: Staring HTTP server
I (3757) wifi: rsn valid: gcipher=3 ucipher=3 akm=5

I (3857) wifi: rsn valid: gcipher=3 ucipher=3 akm=5

I (3957) wifi: rsn valid: gcipher=3 ucipher=3 akm=5

(repeat forever)
```

evansj, 2020-07-15:
Actually it looks like `homekit_run_server: Staring HTTP server` is a success message. It starts a server listening on port 5556. I was able to find the camera in the Home app (default password: all 1s)

closed 2020-09-14 by maximkulkin