maximkulkin/esp32-homekit-camera #75: missing menu option

mentaluproar, 2020-08-01:
I've followed the build instructions in the wiki, but am stuck due to a missing menu option.  
"Support array rtc_gpio_desc for ESP32" is missing.  I can't find it anywhere.  I tried cloning the git again, copy-pasting the instructions, but it's still missing.

cgmckeever, 2020-08-01:
Search the closed issues, someone pointed out a way to `search` the menu over navigating it. I dont think it was this particular menu item, but it helped me.

peros550, 2020-08-01:
I had the same issue, I just skip that option and it compiled successfully

tiagokall, 2020-09-20:
If you type "/" you can search for a keyword, it might help!

maximkulkin, 2020-09-20:
@tiagokall I think the issue is that this option appeared only after particular ESP-IDF version and original author is using version prior to that. Just checked ESP-IDF master and the option is there, exactly where documentation says it is.

sylwesterr, 2020-10-09:
Agree with Maxim, I just tried to configure and was also missing this option and could not compile, but when I installed latest IDF even the menuconfig layout changed (from entered to left-aligned) and the option appeared exactly in the place described in installation instruction.

tophneal, 2023-02-02:
I'm running across this same issue with idf v5. I know that rtc_gpio_desc has been changed to rtc_io_desc, per the changelogs, but I cannot find rtc_io_desc in menuconfig, even when using the "/" shortcut to search. Has this been moved to a submodule, or is it safe to skip this configuration change and proceed with the rest?