maximkulkin/esp32-homekit-camera #76: which IDF version do you use for ESP32-S based modules?

mentaluproar, 2020-08-01:
The documentation says to use a branch, cf056a7, that does not exist.  I'm supposedly not supposed to use the stable branch on this chip.  How are you guys building this?

cgmckeever, 2020-08-01:
This commit ? https://github.com/espressif/esp-idf/commit/cf056a7d0b90261923b8207f21dc270313b67456

mentaluproar, 2020-08-01:
git clone -b cf056a7 --recursive https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'...
fatal: Remote branch cf056a7 not found in upstream origin

that's what I'm getting

cgmckeever, 2020-08-01:
```
git clone git@github.com:espressif/esp-idf.git
cd esp-idf
git checkout cf056a7d0b90261923b8207f21dc270313b67456
```

work off that commit for whatever you need to do

hire-vladimir, 2020-08-01:
i used latest (master branch) without issues

mentaluproar, 2020-08-01:
> ```
> git clone git@github.com:espressif/esp-idf.git
> cd esp-idf
> git checkout cf056a7d0b90261923b8207f21dc270313b67456
> ```
> 
> work off that commit for whatever you need to do

git clone git@github.com:espressif/esp-idf.git
Cloning into 'esp-idf'...
The authenticity of host 'github.com (140.82.113.3)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com,140.82.113.3' (RSA) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

WTH?  Trying the latest master.