maximkulkin/esp32-homekit-camera #80: CMakeLists.txt not found in project directory

Bernhard5, 2020-10-07:
Hallo,

ich habe ein problem beim configurieren. Wenn ich make nutze bekomme ich die Fehlermeldung, dass der Befehl nicht existiert. Wenn ich idf.py menuconfig nutze, wird mir angezeigt CMakeLists.txt existiert nicht.

Ich benutze die neuste ESP-IDF Version unter Windows mit Python 3.7.8 .

Kann mir wer helfen?

Hennoloe, 2020-10-08:
Hi,
I also got the same problem, so i was trying to do it with the cmakelists.txt data from components/esp32-camera.

Told me my CMake version could be to "new"

CMake Error at CMakeLists.txt:31 (register_component):
  Unknown CMake command "register_component".

aoimasu, 2020-11-16:
Please use `make menuconfig` instead of `idf.py menuconfig`

Eistee82, 2020-11-19:
I got this:
`$ make menuconfig`
`Makefile:5: /make/project.mk: No such file or directory`
`make: *** Keine Regel, um „/make/project.mk“ zu erstellen.  Schluss.`

gautam-dev-maker, 2021-01-20:
This issue has been solved in this [pull request](https://github.com/maximkulkin/esp32-homekit-camera/pull/83)