maximkulkin/esp32-homekit-camera #85: How change image resolusion?

feanor413, 2021-02-08:
Hello, how change image resolution?

okanduzyel, 2021-03-13:
config.h

VIDEO_IMAGE_SCALE_DENOM 4

It is the divider of the main resolution that is  640x480. 

VIDEO_IMAGE_SCALE_DENOM 4 -> 160x120
VIDEO_IMAGE_SCALE_DENOM 2 -> 320x240

49adrien, 2021-04-01:
Hello,

When i try to scale the resolution, there is CPU error ?

Thanks

`E (88243) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (88243) task_wdt:  - IDLE0 (CPU 0)
E (88243) task_wdt: Tasks currently running:
E (88243) task_wdt: CPU 0: Camera Stream
E (88243) task_wdt: CPU 1: IDLE1
E (88243) task_wdt: Print CPU 0 (current core) backtrace`

okanduzyel, 2021-04-01:
Disable the WDT.

49adrien, 2021-04-01:
Where is the WDT function ?

49adrien, 2021-04-01:
I don't find this option, where is she ?

okanduzyel, 2021-04-01:
JPEG output size: 640x480 is too big. try to 320x240