maximkulkin/esp32-homekit-camera #92: Config no esp32 homekit camera

ghira89, 2021-09-07:
Sorry for my language 

But not ricognition esp32 homekit camera [image]

Seguendo la guida tutto funziona ma non riesco a trovare il passaggio dove chiede di cambiare i parametri della esp32 homekit camera !

Help 

Thx..!

ghira89, 2021-09-09:
[image]

andrewsampaio, 2021-10-12:
If you try to compile anyways, it throws an error:
In file included from  ../components/homekit/src/crypto.c:13: ../components/homekit/src/port.h:19:10: fatal error: esp_spi_flash.h: No such file or directory
 #include <esp_spi_flash.h>
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

@maximkulkin, please help us. I have ESP-IDF on Windows, just FYI.

steini72, 2021-11-15:
[image]
ESP-IDF on Windows 4.3.1 (mingw32)
Toolchain version: esp-2021r1
Compiler version: 8.4.0