maximkulkin/esp32-homekit-camera #94: No Live Stream

randomiq, 2021-11-12:
Recently gave this project a try and so may have made a few incorrect assumptions.  Ultimately the cameras boot up, are added to homekit and show the 10 sec previews
[image]

However, when I click on a preview to see a live stream, I get just a rotating swirly thing and just the first preview frame.
[image]

I've attached a complete log of the serial debug from boot up, adding to homekit, to live stream, to disconnect.  Nothing jumps out to me.

I'm using the espressif v3.3.5 release branch as the latest master wasn't compiling.  Should I use a newer version?
[esp32-homekit-camera-01.log](https://github.com/maximkulkin/esp32-homekit-camera/files/7524534/esp32-homekit-camera-01.log)

steini72, 2021-11-15:
The same problem
ESP-IDF on Windows 4.3.1 (mingw32)
Toolchain version: esp-2021r1

steini72, 2021-11-15:
[image]

works

steini72, 2021-11-15:
> 
> 
> [image]
> 
> works no

rcarmo, 2022-01-29:
Same problem here. Not sure if the panic handler does anything, TBH, and I've looked at #84, reflashed, and still no live stream.

Anyone got a confirmed fix?

phuzybuny, 2022-05-26:
Toggling the panic handler did not fix the issue for me.

tema-mazy, 2023-02-09:
got the same issue

`>>> HomeKit: [Client 1] Get Characteristics
>>> HomeKit: [Client 1] Requested characteristic info for 1.12 ("Streaming Status")
>>> HomeKit: [Client 1] Update Characteristics
>>> HomeKit: [Client 1] Updating characteristic 1.13 ("Setup Endpoints") with TLV:
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x401d1e4d  PS      : 0x00060930  A0      : 0x800dbe8f  A1      : 0x3ffc34e0  
0x401d1e4d: tlv_get_value at /Users/mazy/Dev/dev/esp32/esp32-homekit-camera/components/homekit/src/tlv.c:102

A2      : 0x40000de8  A3      : 0x00000001  A4      : 0x3f4113c8  A5      : 0x3ffbc07c  
A6      : 0x0000000f  A7      : 0xff000000  A8      : 0x00000014  A9      : 0x3ffc34c0  
A10     : 0x3ffe2418  A11     : 0x000000bc  A12     : 0x3ffb49e0  A13     : 0x00060523  
A14     : 0x00060520  A15     : 0x00000001  SAR     : 0x0000001c  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000dec  LBEG    : 0x4008b72e  LEND    : 0x4008b739  LCOUNT  : 0x00000000  
0x4008b72e: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150

0x4008b739: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace: 0x401d1e4a:0x3ffc34e0 0x400dbe8c:0x3ffc3500 0x400ddb6d:0x3ffc3530 0x400e0e33:0x3ffc3560 0x400e0fef:0x3ffc35d0 0x400e277d:0x3ffc3610 0x400f967d:0x3ffc3630 0x400dede6:0x3ffc3690 0x400e1433:0x3ffc36c0 0x400e2909:0x3ffc3700 0x40091dd9:0x3ffc37c0
0x401d1e4a: tlv_get_value at /Users/mazy/Dev/dev/esp32/esp32-homekit-camera/components/homekit/src/tlv.c:101

0x400dbe8c: camera_setup_endpoints_set at /Users/mazy/Dev/dev/esp32/esp32-homekit-camera/main/accessory.c:212

0x400ddb6d: homekit_characteristic_default_setter_ex at /Users/mazy/Dev/dev/esp32/esp32-homekit-camera/components/homekit/src/accessories.c:411

0x400e0e33: process_characteristics_update at /Users/mazy/Dev/dev/esp32/esp32-homekit-camera/components/homekit/src/server.c:3059
`

or

`I (8985) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 1] Update Characteristics
>>> HomeKit: [Client 1] Updating characteristic 1.13 ("Setup Endpoints") with TLV:
E (9005) esp32_camera: Failed to setup endpoints: no session ID field
>>> HomeKit: [Client 1] Update Characteristics
>>> HomeKit: [Client 1] Subscribed to notifications of characteristic 1.16 ("Volume")
>>> HomeKit: [Client 1] Subscribed to notifications of characteristic 1.17 ("Mute")
>>> HomeKit: [Client 1] Get Characteristics
>>> HomeKit: [Client 1] Requested characteristic info for 1.13 ("Setup Endpoints")
I (9045) esp32_camera: Creating setup endpoints response
>>> HomeKit: [Client 1] Update Characteristics
>>> HomeKit: [Client 1] Updating characteristic 1.14 ("Selected RTP Stream Configuration") with TLV:
>>> HomeKit: [Client 1] Resource
`

than-sad, 2023-02-26:
I have the same problem. `Core  0 panic'ed (LoadProhibited). Exception was unhandled.` when click on camera to live streaming.

barisonal, 2023-03-23:
> I have the same problem. `Core 0 panic'ed (LoadProhibited). Exception was unhandled.` when click on camera to live streaming.

Hi again! :) Did you find a solution for it?

than-sad, 2023-03-24:
Hi! no, but I read on internet guru meditation errors can be related to not enough power. In my case I have some Brownout detector was triggered resets and I think can be the reason of panics.

The no session ID I still don't know why is happening.

Anyway I moved to another project without Homekit but streaming at 1600x1200 with good quality and around 5 frames/second. https://github.com/s60sc/ESP32-CAM_MJPEG2SD

barisonal, 2023-03-24:
> Hi! no, but I read on internet guru meditation errors can be related to not enough power. In my case I have some Brownout detector was triggered resets and I think can be the reason of panics.
> 
> The no session ID I still don't know why is happening.
> 
> Anyway I moved to another project without Homekit but streaming at 1600x1200 with good quality and around 5 frames/second, movement detection, saving videos in to SD, FTP recorded videos upload and email notifications. https://github.com/s60sc/ESP32-CAM_MJPEG2SD

Wow! Sounds great! Actually I strongly want to use Homekit compatible one, because I have set up everything in my home on Homekit. 

But I have check it out and got an idea! This alternative you shared works with MQTT which means it can be integrated to Homekit via Home Assistant. I also use Home Assistant to manage some smart devices in Homekit. Homekit integration works very well in Home Assistant.

than-sad, 2023-03-24:
If you have Home Assistant I recommend you to try it. I have also all in home kit without Home Assistant but 320x240 vs 1600x1200 with more frame rate is another world... In fact I have two esp32-cam one for my room with home kit and another one in the entrance with ESP32-CAM_MJPEG2SD

barisonal, 2023-03-24:
Yes I am going to try it. 

By the way, I recommend you to use Home Assistant Bridge in your Homekit environment. You will have a flexibility for devices even they are incompatible with Homekit.

barisonal, 2023-03-24:
Okay, I tested. It works very well but I'm gonna correct myself! I read it only sends `Record On/Off` and `Motion On/Off` messages. But I also tried to add various camera integrations in Home Assistant.  But bad news. So we can not use it in Home Assistant and of course can not use in Homekit too :(

rcarmo, 2023-03-24:
My cameras with this project can live stream to my phone with iOS 16.3.1. They didn’t a while back, and the stream quality is still… sub-par, but I just noticed they are indeed “working”, including on my Apple TV - the stream drops off every now and then, but recovers after a few seconds (I suppose it’s the panic handler)so something changed…

barisonal, 2023-03-24:
> My cameras with this project can live stream to my phone with iOS 16.3.1. They didn’t a while back, and the stream quality is still… sub-par, but I just noticed they are indeed “working”, including [...]

Yes @rcarmo, it works for us too but with various symptoms or problems. For me & @than-sad, it gets snapshots periodically, that's fine. But it crashes (#111) when we try to switch live mode. 

By using [Homekidd version](https://github.com/HomeKidd) (which has been forked from this project) we can get livestream but with low resolution (320x240). (For me, sometimes it crashes after 3rd or 4th live trigger)

rcarmo, 2023-03-24:
Maybe it wasn’t clear - it recently started working better for me. I will have a look at that fork, it’s 5 years old now, and I don’t have spare hardware to test, but it’s intriguing… I also need a “pure” HomeKit-based solution.