maximkulkin/esp32-homekit-camera #96: make menuconfig does not work

stoex, 2022-03-05:
Hey there, I have a problem with the step "Configuration".

Why doesn't it work?

[image]

stoex, 2022-03-05:
It says that "make" is written false or can't be found

zackdvd, 2022-03-26:
You need to set up your compiler / build tools. In Debian its build-essential, no idea in windows. Have a look here: 
https://github.com/espressif/esp-idf

camelbo, 2022-05-19:
Hello,

as i am playing since 2 weeks to get this run, here my findings:
If you install ESP-IDF on Windows with the esp-idf installer, you have to 

>>>>>>>>>>>>>>>replace "make" with "idf.py"<<<<<<<<<<<<<<<<<

then you are able to idf.py menuconfig and so on.

Here are my commands to apply the package with ESP-IDF 4.4 Powershell:

git clone https://github.com/maximkulkin/esp32-homekit-camera.git
cd esp32-homekit-camera
git submodule update --init --recursive
idf.py menuconfig
git apply --directory="components/esp32-camera" esp32-camera.patch
idf.py all
idf.py erase-flash
idf.py flash monitor

loenrrafael, 2022-05-31:
What about this one?

PS C:\esp32-homekit-camera> idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp32-homekit-camera\build
Executing "ninja all"...
[1/71] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\esp32-homekit-camera\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/esp32-homekit-camera/build/bootloader/bootloader.bin"
Bootloader binary size 0x62b0 bytes. 0xd50 bytes (12%) free.
[6/69] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/twi.c.obj
FAILED: esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/driver/twi.c.obj
ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/esp32-camera/driver/include -I../components/esp32-camera/conversions/include -I../components/esp32-camera/driver/private_include -I../components/esp32-camera/sensors/private_include -I../components/esp32-camera/conversions/private_include -IC:/Espressif/frameworks/esp-idf-v4.4/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include/esp_additions/freertos -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include/esp_additions -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4/components/heap/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/log/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/port/esp32/include/arch -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/public_compat -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/driver/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/driver/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_pm/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_ringbuf/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/efuse/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/efuse/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/vfs/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_wifi/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_event/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_eth/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/tcpip_adapter/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_phy/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_phy/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_ipc/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/app_trace/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_timer/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/nvs_flash/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/spi_flash/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/mbedtls/port/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/mbedtls/mbedtls/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/mbedtls/esp_crt_bundle/include -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Og -fmacro-prefix-map=C:/esp32-homekit-camera=. -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v4.4=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switc [... cut ...]
../components/esp32-camera/driver/twi.c:61:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
                        ^~~~~~~~~~~~~
                        rtc_io_desc
../components/esp32-camera/driver/twi.c:61:24: note: each undeclared identifier is reported only once for each function it appears in
[11/69] Building C object esp-idf/libjpeg/CMakeFiles/__idf_libjpeg.dir/jpeg-6b/jquant2.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
PS C:\esp32-homekit-camera>

tasmotari, 2023-02-18:
> replace "make" with "idf.py"

Unbelievable! I was wasting my time with wrong instruction...

than-sad, 2023-03-10:
@leonrrafael Looks like you forget to activate the RTC option in menuconfig, check de readme to found it.

`Component config
Driver configuration
RTCIO configuration
Support array rtc_gpio_desc for ESP32 = check`