maximkulkin/esp32-homekit-camera #99: Failed to dump SPR public key (code -199)

blaisebarrette, 2022-05-22:
It has been quite a journey to have the code compiled and flashed to the espCAM.
It is now on there and the Home app sees the camera but when I try to connect to it It tries for a few second and it fails with a "Unknown error occurred" message.
In the serial monitor it looks like that:

`>>> HomeKit: [Client 1] Got new client connection from 192.168.1.126
>>> HomeKit: [Client 1] Pair Setup Step 1/3
!!! HomeKit: [Client 1] Failed to dump SPR public key (code -199)
>>> HomeKit: [Client 1] Closing client connection from 192.168.1.126`

Any one have come across this problem?

Thanks! 

- Blaise

camelbo, 2022-05-22:
Hello,

at issues, there are 2 registers. Open issues and closed issues. I asked the same question yesterday.

solution: git submodule update --remote --merge

blaisebarrette, 2022-05-22:
Thanks for the quick reply!

It definitely changed something. It went a little bit further but instead of a "Unknown error occurred" message, It said that the accessory was not compatible and now it refuses to see it.

>>> HomeKit: Starting server
I (7624) esp32_camera: Initializing streaming
>>> HomeKit: Using existing accessory ID: CD:4F:0F:0D:50:45
>>> HomeKit: Found admin pairing with C3127C22-2728-4633-8CCF-1AE25D762D1D, disabling pair setup
I (7634) esp32_camera: Streaming initialized
>>> HomeKit: Configuring mDNS

blaisebarrette, 2022-05-23:
I flashed a different espCAM and this time it's working.
The result is rather disappointing though.
I only get a new image every 10 seconds and if I select the camera for full screen, it shuts down. 
Any chance I can get a better result from this project?
Some settings aren't set properly maybe?

Thanks! - Blaise

loenrrafael, 2022-06-10:
Similar here:
E (13886) esp32_camera: Failed to setup endpoints: no session ID field

No video stream

AchimPieters, 2023-02-09:
solution: git submodule update --remote --merge