maximkulkin/esp32-homekit-camera #100: Failed to setup endpoints: no session ID field

phuzybuny, 2022-05-28:
Previews are working but live video is not. The issue does not appear to be watchdog related.

```
>>> HomeKit: [Client 1] Updating characteristic 1.13 ("Setup Endpoints") with TLV:
E (34245) esp32_camera: Failed to setup endpoints: no session ID field
```

Looks like `v` is NULL because `request->head->type` is 20. 
```
v = tlv_get_value(request, 1);
```

Removing the validation just results in the session ID being an invalid size.

```
E (21525) esp32_camera: Failed to setup endpoints: session ID field has invalid size (1073626880)
```

_Originally posted by @phuzybuny in https://github.com/maximkulkin/esp32-homekit-camera/issues/94#issuecomment-1138691192_

loenrrafael, 2022-06-12:
Similar issue here. Any advice?

daveholiday, 2022-06-21:
I'm also having this problem.  Has anyone made progress?

leonrrafael, 2022-06-21:
No luck yet.

rcarmo, 2022-07-02:
No luck here either.

CP02A, 2022-09-05:
I have the same problem - has anyone succeeded at fixing the problem?

leonrrafael, 2022-09-05:
Nope. I think the developer left the project.

Hellangel1987, 2023-02-09:
im now also facing this issue

[image]

E (26190) esp32_camera: Failed to setup endpoints: no session ID field