maximkulkin/esp32-homekit-camera #102: Build with errors

pawelosin, 2022-07-23:
Toolchain path: /Users/XXXXXXXXi/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/XXXXXXXX/esp32/esp-idf/requirements.txt are satisfied.
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/component.mk:86: warning: undefined variable `IDF_TARGET'
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/component.mk:88: warning: undefined variable `IDF_TARGET'
CC build/homekit/src/debug.o
In file included from /Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:4:0:
/Users/XXXXXXXXesp32-homekit-camera/components/homekit/src/debug.h:33:23: error: unknown type name 'uint8_t'
 char *data_to_stringv(uint8_t n, const byte **datas, size_t *sizes);
                       ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.h:36:23: error: unknown type name 'uint8_t'
 char *text_to_stringv(uint8_t n, const byte **datas, size_t *sizes);
                       ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.h:39:25: error: unknown type name 'uint8_t'
 char *binary_to_stringv(uint8_t n, const byte **datas, size_t *sizes);
                         ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c: In function 'data_to_string':
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:8:12: error: implicit declaration of function 'data_to_stringv' [-Werror=implicit-function-declaration]
     return data_to_stringv(1, (const byte*[]){ data }, (size_t[]){ size });
            ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:8:12: warning: return makes pointer from integer without a cast [-Wint-conversion]
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c: In function 'text_to_string':
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:13:12: error: implicit declaration of function 'text_to_stringv' [-Werror=implicit-function-declaration]
     return text_to_stringv(1, (const byte*[]){ data }, (size_t[]){ size });
            ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:13:12: warning: return makes pointer from integer without a cast [-Wint-conversion]
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c: In function 'binary_to_string':
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:18:12: error: implicit declaration of function 'binary_to_stringv' [-Werror=implicit-function-declaration]
     return binary_to_stringv(1, (const byte*[]){ data }, (size_t[]){ size });
            ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:18:12: warning: return makes pointer from integer without a cast [-Wint-conversion]
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c: At top level:
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:22:24: error: unknown type name 'uint8_t'
 bool is_human_readable(uint8_t n, const byte **datas, size_t *sizes) {
                        ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:41:23: error: unknown type name 'uint8_t'
 char *text_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                       ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:79:25: error: unknown type name 'uint8_t'
 char *binary_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                         ^
/Users/XXXXXXXX/esp32-homekit-camera/components/homekit/src/debug.c:109:23: error: unknown type name 'uint8_t'
 char *data_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                       ^
cc1: some warnings being treated as errors
make[1]: *** [src/debug.o] Error 1
make: *** [component-homekit-build] Error 2

mkruselhind, 2022-12-03:
Have the same error. Any ideas so far ?

than-sad, 2023-02-22:
Same error for me. Please any help?

than-sad, 2023-02-22:
Hi, I added `#include <stdint.h>` as the first include inside /esp32-homekit-camera/components/homekit/src/debug.h

Now compilation goes further until error:
```
.../esp32-homekit-camera/main/app_main.c:49:21: error: implicit declaration of function 'esp_event_loop_init' [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
                     ^
/Users/than_sad/esp32/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);   
```

If someone can help will be really apreciated.

than-sad, 2023-02-23:
Hi, tried in a old Mac with High Sierra with same result. This is really frustrating.

than-sad, 2023-02-26:
Ok I found in this git files changes just adding `#include <esp_event_loop.h>` after `#include <esp_event.h>` on file ./main/app_main.c solved the problem and allowed me to compile and flash it correctly FINALLY!!!